realsenseai/librealsense #8743: RGB infrared frame gets messed up when requesting both the IR frames

ShravyaB, 2021-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense D415 | 
| Firmware Version                   |  | 
| Operating System & Version |   MacOS | 
| Kernel Version (Linux Only)    |  N/A | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.36.0 }                          | 
| Language                            |  C/C++                          | 
| Segment			| Others  | 

### Issue Description
[image]

When enabling both IR frames and one in RGB/BGR I see the following corruption. When both are Y8 or Y16, I don't see this issue. If I only enable one of the IR cameras I don't see this issue.
Same issue is reproduced in code

MartyG-RealSense, 2021-04-05:
Hi @ShravyaB  I have not got access to a Mac to test on.  It has long been my experience though that having **color from left infrared** (RGB8) enabled on the 'Infrared' stream prevents activation of the stereo module if 'Infrared 2' is also enabled.

[image]

If you have previously been able to successfully stream RGB8 and Y8 in the Viewer together on your Mac, is the problem resolved if you reset the Viewer to its default settings, please?  This procedure can be performed with the following steps:

1.  Left-click on the gear-wheel icon in the top corner of the Viewer and select the **Settings** option from its menu to open the Settings interface of the Viewer.

[image]

2.  Left-click on the **General** tab of the Settings window to select that section of options and then left-click on **Restore Defaults**, and finally **Apply** to reset the Viewer.

[image]

ShravyaB, 2021-04-05:
Well, I have not tried restore defaults but I did try hardware reset and it didn't change. Currently I don't have the Realsense with me so I'll try this approach tomorrow and let you know.

MartyG-RealSense, 2021-04-05:
They are different types of reset.  Hardware Reset resets the **camera hardware**, whilst the Restore Defaults option in the Settings interface affects the **Viewer software**.   Yes, please do update here when you are able to test.  Thanks!

ShravyaB, 2021-04-06:
Thank you for responding so promptly. I have tried resetting it but the output did not change.
As I mentioned earlier, even without the viewer (when I run it in my code, I see the same issue). To check if it a MacOS issue or a connector issue, I tried using this device with a Windows 10 system and got the following error:

Invalid Value in rs2_open_multiple(sensor:000001C9921B33D0, profiles:000001C9988CC9E8, count:2): 

Failed to resolve the request: 
    Format: RGB8, width: 1280, height: 720
    Format: Y8, width: 1280, height: 720

Into:
    Formats: 
     UYVY
     Y8I

Trying to use the solution from this link: https://www.intel.in/content/www/in/en/support/articles/000028857/emerging-technologies/intel-realsense-technology.html

Let me know your thoughts

ShravyaB, 2021-04-06:
Even a firmware update did not resolve the issue. Not sure what changed

MartyG-RealSense, 2021-04-07:
I conducted further research but was unable to find a solution.  As mentioned earlier, it is my own experience on non-Mac computers that Infrared 2 cannot stream when Infrared 1 is using RGB8 color from left infrared.

Is it possible for you to use Y8 or Y16 on infrared 1 and RGB for the color?

I would also highlight  that firmware version **5.12.7.100** is s better choice for SDK 2.36.0.   From firmware 5.12.8.200 onwards there is the requirement to use at least SDK 2.39.0 due to internal firmware changes.  So it may be worth downgrading the firmware back to 5.12.7.100.

ShravyaB, 2021-04-07:
Makes sense. I am facing this issue ONLY when trying to stream RGB8 or any other color format together with Y8.

But the crazy thing is that it was working just minutes before so not sure what changed

MartyG-RealSense, 2021-04-07:
You could try uninstalling the RealSense SDK and installing it again.  If the SDK is otherwise working normally for you though then it may not be worth the risk of breaking what is currently working.

closed 2021-04-19 by MartyG-RealSense