realsenseai/librealsense #8744: How to see if frames got dropped?

snopytas, 2021-04-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2                          | 
| Language                            |  C/C++                       | 

### Issue Description
Hi, 

I'm receiving frames via `pipe.poll_for_frames()` . I would like to see (or get notified) if frames get dropped because they couldn't get polled fast enough. 

I'm already saving debug and info messages with `rs2::log_to_file(RS2_LOG_SEVERITY_ALL,"logfile.log")` but I can't see anything in the logfile which indicates that a frame got dropped, even if I artificially delay the processing, e.g.

```
rs2::log_to_file(RS2_LOG_SEVERITY_ALL,"LOG.log");
rs2::frameset data;
while(true) {
    if (pipe.poll_for_frames(&data)) {
        ... do some stuff ...
        ...wait for 500ms ...
    }  
}
```

Where can I see if a frame got dropped?

Thanks and best regards

MartyG-RealSense, 2021-04-05:
Hi @snopytas   There is a discussion in the link below with information about how frame drops are calculated in the SDK.

https://github.com/IntelRealSense/librealsense/issues/7488#issuecomment-704124850

That discussion points to a group of C++ lines in the SDK that are responsible for frame drop calculation.  This will hopefully be a helpful reference that you can adapt for your own project.

https://github.com/IntelRealSense/librealsense/blob/master/common/output-model.cpp#L1099-L1108

snopytas, 2021-04-06:
Thank you @MartyG-RealSense. The [documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md#frame-memory-management) says that a frame drop would be visible in the debug log, but I don't see it. Maybe it could be due to the fact that for the moment I'm playing a recorded file (Homeoffice) instead of using the real camera.

Anyway, I was able to adapt the code you referenced and this is sufficient for my purposes. So this issue is solved for me. Thank you very much!

MartyG-RealSense, 2021-04-06:
You are very welcome @snopytas   :)   If you are satisifed with the outcome of the case, please feel free to close it with the **Close Issue** button beneath the comment writing box.  Thanks again!

closed 2021-04-07 by snopytas