realsenseai/librealsense #8745: Access Depthmap Frame Intrinsics from Viewer ROS .bag Files

jrder, 2021-04-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.12.10.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  n/a                                    | 
| Platform                                 | PC  |
| SDK Version                            | n/a                   | 
| Language                            |  MATLAB 2020b                         | 
| Segment			| Other                        | 

### Issue Description
I'm using the Viewer software to record and save depthmaps from my D435 to ROS .bag files. I would like to access the camera frame intriniscs from the ROS .bag files.

I'm using the depthmap decimation post-processing filter, which changes the camera intrinsics for the depthmaps. This makes them different than the calibrated instrincs available via the SDK and calibration table in the Viewer.

I'm using the frame intrinsics to perform deprojection on processed depthmaps to pointclouds {x,y,z} data. Because I'm processing my depthmaps I'm unable to just read the pointclouds from the device for my application.

So far, I've read the topics and messages in the ROS .bag file in MATLAB 2020b, trying to identify the frame intrinsics.

Thank in advance.

MartyG-RealSense, 2021-04-06:
Hi @jrder   Is the solution that you require dependent on MATLAB, please?  If it is possible for you to use Python, the link below has a script for extracting the intrinsics of each frame from a bag.

https://github.com/IntelRealSense/librealsense/issues/7309

jrder, 2021-04-06:
Hi @MartyG-RealSense, I have to use MATLAB, my depthmap processing is done there. 

However, the thread you linked is exactly what I need. I didn't know that you could read the bag file with the SDK. I should be able to replicate the process in MATLAB.

Thank you.

MartyG-RealSense, 2021-04-06:
It's great to hear that the link was helpful - thanks very much for the update!

jrder, 2021-04-08:
Hi @MartyG-RealSense, I've been able to write a function similar to the linked issue in MATLAB. However, the playback fails when waiting for the 16th frame of a 48 second recording. 

I've read other similar issues, where they use a playback object to set_real_time to false, and ensure that the repeat_playback option for the configuration is set to false.

Any further help would be appreciated.

Below is my MATLAB code so far.

```
function [frameset] = rs2_readRosbag(filename)
%% Read ROS .bag file created from Intel Realsense Viewer Recording
% by jrder
% date April 6, 2021
% inputs: ROS .bag filename / absolute file path 
% outputs: array of frame objects

arguments
  filename
end

%% initializations for replaying bag file
  % configuration 
  cfg = realsense.config();
  % pipeline
  pipe = realsense.pipeline();
  % frameset empty array 
  frameset = [];
  % frameset counter
  counter = 0;
  % frameset while loop condition 
  frames_exist = true;

%% replay bag file
  % replay file through configuration
  cfg.enable_device_from_file(filename,false);
  % stream bag file through pipeline
  profile = pipe.start(cfg);
  % create playback object
  playback = profile.get_device().as('playback');
  % set real time to false to catch all frames
  playback.set_real_time(false);
  
%% count number of frames and append to frameset array
  % while loop through frames until error occurs
  while frames_exist
    % attempt to collect frame
    try
      % read frame
      fs = pipe.wait_for_frames(); <--------------------- fails here after counter reaches 15
      % append frame to frameset array 
      frameset = [frameset fs];
      % increment counter
      counter = counter + 1;
    catch
      % when error occurred, write text
      fprintf('number of frames %2.0f\n',counter);
      % set while loop condition to false
      frames_exist = false;
      % exit loop
      break
    end
  end
  % stop bag file replay 
  pipe.stop(); 

end
```

MartyG-RealSense, 2021-04-08:
If the above code is Python then that would explain the program stopping after the 15th frame.  This is a known effect of using an **append** in Python to save frames into a collection.

https://github.com/IntelRealSense/librealsense/issues/946

The link below provides details of two different methods that have been used in the past to avoid this limitation.

https://github.com/IntelRealSense/librealsense/issues/8652#issuecomment-804993692

jrder, 2021-04-08:
Hi @MartyG-RealSense, thank you for your help! 

I wasn't able to get the .keep() method to work in MATLAB, it was recognized as a function but caused pipeline.wait_for_frames() to timeout on the first frameset. However, I was able to find a workaround suitable for my application. 

I wanted to get the frame intrinsic for the depth maps recorded in the bag file. Instead of collecting the framesets, then processing them together, I just processed them individually. This included querying the depth frames for their intrinsic.

Notably, I was able to read the ROS bag file using the MATLAB ROS bag Toolbox for the all streams (color, infrared, and depth map). I found this faster, however, with this approach I'm unable to use the SDK to process and query the individual frame objects. Also, I was unable to locate the frame intrinsic in this data format.

Thanks again for all your help.

closed 2021-04-08 by jrder

MartyG-RealSense, 2021-04-09:
Thanks very much @jrder for sharing your solution  :)