realsenseai/librealsense #8746: Gyroscope coordinates in c #

anomal3, 2021-04-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   {Windows 10 LTSC  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.43.0}                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
Good afternoon friends. I found a great example that I need, but I cannot implement it in c #
 [link to Exaple!](https://dev.intelrealsense.com/docs/rs-motion) and [link to source code C++](https://github.com/GruffyPuffy/imutest/blob/master/imutest.cpp)

[image]

I do not need such a graph in 3d
I just need the position coordinates.

Has anyone already done this? Pulling out the coordinates?

I'm talking about it
[image]

If you know how to implement in c # please share an example with me

MartyG-RealSense, 2021-04-06:
Hi @anomal3   There is not a lot of information available for accessing IMU streams with C#.  There was a RealSense user who did so by using **callbacks**.

https://support.intelrealsense.com/hc/en-us/community/posts/360044035053-C-Time-out-with-D435i-IMU-data-enabled
https://support.intelrealsense.com/hc/en-us/community/posts/360044035053/comments/360011430533

An alternative approach suggested was to access the information through the RealSense SDK's **Low-Level API** (accessing the camera hardware directly).

https://github.com/IntelRealSense/librealsense/issues/2996#issuecomment-451068939

anomal3, 2021-04-06:
> Hi @anomal3 There is not a lot of information available for accessing IMU streams with C#. There was a RealSense user who did so by using **callbacks**.
> 
> https://support.intelrealsense.com/hc/en-us/community/posts/360044035053-C-Time-out-with-D435i-IMU-data-enabled
> https://support.intelrealsense.com/hc/en-us/community/posts/360044035053/comments/360011430533
> 
> An alternative approach suggested was to access the information through the RealSense SDK's **Low-Level API** (accessing the camera hardware directly).
> 
> [#2996 (comment)](https://github.com/IntelRealSense/librealsense/issues/2996#issuecomment-451068939)

thank you very much.
This is what I need. There is only a small BUT

```csharp
var accelFrame = frameset.FirstOrDefault<Frame>(Stream.Accel, Format.MotionXyz32f).DisposeWith(frameset);
                    var b = Marshal.PtrToStructure<Vector3>(accelFrame.Data);
```
in win forms Error. *PtrToStructure cannot be used with type arguments*

If you use a console application, everything works fine.

MartyG-RealSense, 2021-04-06:
Thanks very much @anomal3 - I'm pleased that I could help!  Do you require further assistance with this case, please?

anomal3, 2021-04-06:
> Thanks very much @anomal3 - I'm pleased that I could help! Do you require further assistance with this case, please?

Problems with PtrToStructure  only
But in general, everything has been decided!

MartyG-RealSense, 2021-04-06:
Great to hear.  Please feel free to close the case with the **Close Issue** button underneath the comment writing box if you are satisfied with the outcome.  Thanks again!

closed 2021-04-06 by anomal3