realsenseai/librealsense #8750: Camera Frezee

sbilgi, 2021-04-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.12.07 / 5.12.10 | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |  5.9.8 / 5.11                                        | 
| Platform                                 | arm64 / rk3399  |
| SDK Version                            |  2.41                       | 
| Language                            |  C++                       | 

### Issue Description
Hi all,

I tried to capture frame from d430 device. But some camera (not observed every camera) capture frozen image (Sometimes infrared was frozen sometimes depth). I captured only Infrared 1 and depth image (`848x480`). 
I tried stress tool to load on CPU but didn't observe frezing on those cameras. 

I measure CPU usage `%~50` on camera camera capturing thread.

Check dmesg on found in message: `usb 6-1: Process xxx (xxx) called USBDEVFS_CLEAR_HALT for active endpoint 0x83`

Here is two sample camera frezing:

[image]
[image]

MartyG-RealSense, 2021-04-06:
Hi @sbilgi   The link below has a past case where the color stream was freezing during high CPU utilization and causing the **USBDEVFS_CLEAR_HALT for active endpoint** error.  They initially tested on Raspberry Pi 4 and found that reducing resolution downward from 1280x720 depth improved performance.  They were eventually able to achieve long-term stable performance on an RK3399 at 640x480 resolution and 30 FPS.

https://github.com/IntelRealSense/librealsense/issues/5412

Is use of a lower resolution an option for your project, please?

asbilgi, 2021-04-06:
Unfortunately, I can't use it in low resolution. I even want to use 1280x720 at 15 FPS. 
But the stress test didn't generate any error, and when I measure CPU usage it shows %50, I think it's not high.
Is there a root cause for this? or any solution. For example, will it be solved if I turn off my post-processing (currently only depth to disparity and temporal filter)?

sbilgi, 2021-04-06:
@MartyG-RealSense  Unfortunately, I can't use it in low resolution. I even want to use 1280x720 at 15 FPS.
But the stress test didn't generate any error, and when I measure CPU usage it shows %50, I think it's not high.
Is there a root cause for this? or any solution. For example, will it be solved if I turn off my post-processing (currently only depth to disparity and temporal filter)?

MartyG-RealSense, 2021-04-06:
Post-processing does have a processing cost because it is computed on the computer / computing device and not within the camera hardware.

If you built the librealsense SDK from source code with CMake, did you include the **-DCMAKE_BUILD_TYPE=Release** build flag in the CMake build instruction?  This builds librealsense with optimizations.  Without this flag, librealsense may be built in Debug mode which has slower performance.

sbilgi, 2021-04-06:
It's compiled with release flag.
Openmp can solve this problem?

MartyG-RealSense, 2021-04-07:
The CMake build flag documentation states about the **OpenMP** setting (which is set to False by default): "When enabled, YUY to RGB conversion and Depth-Color spatial alignment will take advantage of multiple-cores using OpenMP.  This can reduce latency at expense of greater CPU utilization".

So as you are not using the color stream and are not performing depth-to-color alignment, you are not likely to receive a performance benefit from setting OpenMP to True.

You could temporarily disable infrared streaming to test whether the lag is a consequence of running two simultaneous streams (depth and infrared) on the computing hardware that the camera is attached to.

sbilgi, 2021-04-07:
@MartyG-RealSense  thanks for information about OpenMP.

I noticed that problem is memory usage. Today I tried rs-distance application with stress for memory ( `stress --vm 1 --vm-bytes 1790M --timeout 600` ) It generate same error in `/var/log/kern.log` (I have 2 gb memory on rk3399) for every time. 
You can reproduce error by rs-distance and stress command that I mentioned above.

MartyG-RealSense, 2021-04-08:
If memory usage is the suspected cause, would it be possible to set up a 2 GB **swap file** on your Linux distribution?  This can use storage space as virtual memory when the real memory capacity is used up on the computing device that the camera is attached to.

https://linuxize.com/post/create-a-linux-swap-file/

closed 2021-04-21 by MartyG-RealSense