realsenseai/librealsense #8752: C#: How to save depth frames to 16bit grayscale images

jwolli, 2021-04-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  { Win (10) } |                                    | 
| Platform                                 | { PC } |
| SDK Version                            |  { 2.42.0 }                           | 
| Language                            |  {C# .NET Core 3.1}                          | 

### Issue Description
Hi there!

I am trying to save depth images (grayscale, not colorized) from a D435 to 16bit png-files in C# (.NET Core 3.1).
However I wasn't successful using .NET's Image.Bitmap build-in class for that.
Saving a RGB-image works fine but when I execute the following lines:

```
var streamConfig = new Config();
int width = 848;
int height = 480;
streamConfig.EnableStream(Stream.Depth, width, height, Format.Z16);
streamConfig.EnableStream(Stream.Color, Format.Rgb8);

var pipeline = new Pipeline();
var pipeProfile = pipeline.Start(streamConfig);

using (var frames = pipeline.WaitForFrames())
{
    var depthFrame = frames.DepthFrame.DisposeWith(frames);
    var colorFrame = frames.ColorFrame.DisposeWith(frames);

    var depthImage = DepthFrametoBitmap(depthFrame);
    depthImage.Save(@"C:\src\testImg_depth.png");
}

private Bitmap DepthFrametoBitmap(VideoFrame frame)
{
    Bitmap bitmap = new Bitmap(frame.Width, frame.Height, frame.Stride, PixelFormat.Format16bppGrayScale, frame.Data); 
    return bitmap;
}
```

Unfortunately I couldn't find any other thread regarding this topic.
Is there different approach to export 16bit depth images?

Kind regards
Markus

MartyG-RealSense, 2021-04-06:
Hi @jwolli  I could only find one script, in the link below, for saving a grayscale depth image as a PNG in C#.

https://github.com/IntelRealSense/librealsense/issues/3796

jwolli, 2021-04-06:
Hi @MartyG-RealSense !
Thanks for the quick reply! 
Unfortunately the thread you mentioned does not really help me with this issue.
Although it appears to be grayscale, a colorizer is used and the created image contains three channels. The output format is *PixelFormat.Format24bppRgb* .
What I would like to export is a grayscale image (preferably 16bit) with only one channel.
I tried to use the format *PixelFormat.Format16bppGrayScale* but execution will lead to an error:

> System.Runtime.InteropServices.ExternalException (0x80004005): A generic error occurred in GDI+.
   at System.Drawing.SafeNativeMethods.Gdip.CheckStatus(Int32 status)
   at System.Drawing.Image.Save(String filename, ImageCodecInfo encoder, EncoderParameters encoderParams)
   at System.Drawing.Image.Save(String filename, ImageFormat format)
   at System.Drawing.Image.Save(String filename)
....

Kind regards
Markus

MartyG-RealSense, 2021-04-06:
I recall another case where someone wanted a grayscale image without using the colorizer.  Though the script in the link below is Python, it might provide some ideas.  In that script, the RealSense user converted the depth frame to a grayscale map using **depth_image = cv.convertScaleAbs(depth_image, alpha=0.03)**

[image]

jwolli, 2021-04-06:
Thanks for sharing the link!
I managed to export grayscale depth frames in python using numpy and openCV with the following lines of code:

```python
import pyrealsense2 as rs
import cv2
import numpy as np

pipeline = rs.pipeline()

cfg = rs.config()
cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

profile = pipeline.start(cfg)

frames = pipeline.wait_for_frames()
depth_frame = frames.get_depth_frame()

depth_image = np.asanyarray(depth_frame.get_data())

# save image using openCV
cv2.imwrite("C:\\src\\testImg_depth.png", depth_image)
``` 

Is there a way to achieve a similar result in C#?

MartyG-RealSense, 2021-04-07:
I researched your question carefully but was unable to find an example of C# code for exporting a gray-scale depth image file unfortunately.

MartyG-RealSense, 2021-04-13:
Hi @jwolli  Bearing in mind the above comment, do you require further assistance with this case please?  Thanks!

jwolli, 2021-04-13:
Hi @MartyG-RealSense 
thanks for your time and effort! 
Unfortunately I wasn't able to solve this issue. It seems to me that the pixel format *PixelFormat.Format16bppGrayScale* exists, but somehow cannot be used to export images...
You may close this issue.

MartyG-RealSense, 2021-04-13:
Thanks very much @jwolli for the update.  I'm sorry that you didn't achieve the desired outcome.  As suggested, I will close the case.  Good luck with your work!

closed 2021-04-13 by MartyG-RealSense

asheesh1996, 2021-08-03:
Hi,
If anyone still trying to get it done in c#
```
var streamConfig = new Config();
int width = 848;
int height = 480;
streamConfig.EnableStream(Stream.Depth, width, height, Format.Z16);
streamConfig.EnableStream(Stream.Color, Format.Rgb8);

var pipeline = new Pipeline();
var pipeProfile = pipeline.Start(streamConfig);
using (var frames = pipeline.WaitForFrames())
{
  using (var colorframe = frames.ColorFrame.DisposeWith(frames))
  {
    using (Bitmap bitmap = new Bitmap(colorframe.Width, colorframe.Height, colorframe.Stride, System.Drawing.Imaging.PixelFormat.Format24bppRgb, colorframe.Data))
      {
      using (var memoryStream = new MemoryStream())
        {
        bitmap.Save(memoryStream, System.Drawing.Imaging.ImageFormat.Jpeg);
        }
      }
  }
  using (var depthframe = frames.DepthFrame.DisposeWith(frames))
  {
    byte[] managedArray = new byte[depthframe.DataSize];
    depthframe.CopyTo(managedArray);
    using (var image = SixLabors.ImageSharp.Image.LoadPixelData<SixLabors.ImageSharp.PixelFormats.L16>(managedArray, depthframe.Width, depthframe.Height))
      {
      using (var memoryStream = new MemoryStream())
        {
        image.Save(memoryStream, new PngEncoder());
        }
      }
  }
}

```

MartyG-RealSense, 2021-08-03:
Thanks so much @asheesh1996 for sharing your C# script solution with the RealSense community  :)