realsenseai/librealsense #8756: Align depth to color from saved images

Todibo99, 2021-04-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 / D435 } | 
| Operating System & Version |   {Ubuntu 18}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| Language                            |  {python } | 

### Issue Description
Hi,
I have some saved color and depth images from D415 and D435. I would like to align depth to color. However, the pyrealsense2.align API requires a composite frame as input. I am wondering is there a way to restore frames from saved images or do you have any other suggestions?

Thanks!

MartyG-RealSense, 2021-04-06:
Hi @Todibo99  When a depth frame is saved to PNG, most of the depth information is lost.  It is therefore recommended to save depth to a format that preserves the depth information, such as a **.raw** image.

The link below features a Python script by a RealSense user that took the approach of aligning PNG color frames and an **.npy** file of the depth as an array of scaled matrices.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225

An approach to the composite frame issue may be to manually create a custom frameset using **software_device** or a **custom processing block**.

https://github.com/IntelRealSense/librealsense/issues/5847

Information about siftware_device can be found in the documentation of the SDK's C++ example program for it.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

The number of references available for using software_device with Python are more limited, though the lik below may be helpful.

https://github.com/IntelRealSense/librealsense/issues/4197

Todibo99, 2021-04-10:
thx, i have sorted it out

MartyG-RealSense, 2021-04-10:
Great to hear, @Todibo99  - thanks for the update!

closed 2021-04-16 by MartyG-RealSense