realsenseai/librealsense #8757: Question about frame-archive debug messages

csr-kick, 2021-04-06:
T265 and D435 on RPI4

I have RS2_LOG_SEVERITY_DEBUG set and am getting frames from both the T265 and D435
The console outputs looks like (I have snipped for just pose, but the exact same thing happens for depth):
```
14:31:04,407 DEBUG [3660913424] (archive.cpp:293) CallbackStarted,Pose,32304,DispatchedAt,1617715864407.777832
14:31:04,407 DEBUG [3660913424] (archive.cpp:293) CallbackStarted,Pose,32304,DispatchedAt,1617715864407.924561
14:31:04,407 DEBUG [3660913424] (frame-archive.h:148) CallbackFinished,Pose,32303,DispatchedAt,1617715864407.972412
14:31:04,408 DEBUG [3744840464] (frame-archive.h:148) CallbackFinished,Pose,0,DispatchedAt,1617715864408.260742
14:31:04,408 DEBUG [3744840464] (frame-archive.h:155) Frame Callback [Pose#0] overdue. (Duration: 1617715864408.260742ms, FPS: 200, Max Duration: 6ms)
```

Reading this is seems like the following happens:
1. We register a callback twice
2. Both callbacks return from USB land.
3. Only one actually has data, the other is just zeros (hence the Pose#0 or Depth#0, and a duration equal to now-0)
4. We correctly parse the one with data and it goes on its way, but the one that is all zeros then gets thrown away? 

Why are we creating this frame or registering the callback twice? 
Its clearly two different timestamps, so why the same frame number?
If this is somehow supposed to happen, with empty frames running around doing nothing, why?

RealSenseSupport, 2021-04-08:
@csr-kick Did you get this with your own code? If so, could you please share your code or code snippet.  And also did you get chance to try with SDK examples such as https://github.com/IntelRealSense/librealsense/tree/master/examples/callback to see  if you get this issue also? 

Besides, could you please provide the information below?

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          |

csr-kick, 2021-04-08:
Updated top comment with setup.

Take rs-pose  or rs-depth examples. 

Add in the line:
`rs2::log_to_console(RS2_LOG_SEVERITY_DEBUG);`
as the first line in main.

recompile and run either example.

RealSenseSupport, 2021-04-08:
@csr-kick Could you please provide more detailed information about SDK(librealsense) and Firmware version for both T265 and D435? You can check this with RealSenseViewer if you've installed it. 
[image]

And one more question - did you get such issue with only one camera connected?

csr-kick, 2021-04-08:
As I updated on the first post, it shows this problem on all version at least 2.35 and up (I have not gone lower), with the associated firmwares of those sdk versions. The D435i is anywhere from 5.12.5 to 5.12.12 and the t265 is 2.0.951

It does happen with only one device connected. 

This is a trivial problem for you to verify on your end, grab an RPI4, add the single line to your examples, build, and run.

RealSenseSupport, 2021-04-09:
@csr-kick Thanks for your information! Will have a try on my side.

Nir-Az, 2021-08-31:
Hi @csr-kick ,

Thanks for notifying about this issue (and sorry for the late response).
We investigated the issue and we discovered that the prints were somehow confusing (You are right!).

I will explain:
The duplicated prints of "CallbackStarted" is not cause by duplicated frames or callbacks.
The prints were inside a generic mechanism that served several features, It was printed first at the start of the first callback ,
second at the next processing block that follows it and so on..

The wrong duration time lines were printed probably because the frame was a composite frame  (a combo frame that holds several frames inside)  release which is missing metadata since each frame inside holds different data.

We are refactoring our logs for the frame lifetime currently.
I will update this ticket once we have a merged PR for it.

csr-kick, 2021-09-01:
That's a pretty obscure bug, and I will bet it was not fun to figure out.  I found it while investigating the T265's problem where it will stop sending data to the host computer:

`INFO [2949632528] (tm-device.cpp:1408) T265 FW message: 13042424847758289392: [0x�/2:721] Host not reading - stopping`

See [6362](https://github.com/IntelRealSense/librealsense/issues/6362#issuecomment-637191852)

In searching for a solution to that bug, this error kept appearing. I spoke with some people who were more experienced with USB than I am, and their advice to try and fix the 'Host not Reading' bug was to use the USB async transfer functions to add more than one pending transfer to the transfer ring. 

This was on the theory that one of the two devices, be it the RPI4 or the T265, was periodically missing one side the data transfer request. Based on the message coming from the T265, 'Host not reading', the T265 was probably queueing up the data internally, but the RPI4 was either generating the request too quickly when the T265 is not in a position to receive it, or the request never gets to the T265 in the first place, so the device sits and waits for a transfer request that never arrives while the RPI4 thinks everything is fine. 

By queueing up an additional pending transfer, maybe on a delay, one request should always be ready. The multiple 'callback started/finished' error made it look to me like maybe there WAS an additional transfer queued, and one of the two was returned with no data, then went through your system and printed out the message with all 0 values.

Unfortunately, I stopped there with implementing the multiple transfer idea, as it became clear that we had to move our resources and focus to a solution that was more reliable out-of-the-box. I know you stopped supporting the T265 a long time ago, but it would be such a gift if someone were to be able to solve this particular problem with the device for those of us who still own them. I cannot stress enough how this one single bug doomed this device's sales. My company alone would have purchased hundreds, if not thousands, of T265s had this been fixed. But, as you are going to find out with the depth sensors, no company will base the foundations of their product on a sensor with questionable support and an uncertain future.

Nir-Az, 2021-09-05:
@csr-kick ,Thanks for sharing this information.

Regarding the logs, 
the logs has been refactored and should be much more user friendly now.
PR #9716 .
Will be contained in the next releaser.

I hope it helps.
Good luck.

RealSenseSupport, 2021-09-24:
@csr-kick Did you get chance to try https://github.com/IntelRealSense/librealsense/pull/9716 and any other questions about this? Looking forward to your update. Thanks!

RealSenseSupport, 2021-11-04:
@csr-kick Sorry that we didn't get update from you for weeks. Will close the ticket at this point. Please feel free to create another new ticket if you still have questions or issues. Thanks!

closed 2021-11-04 by RealSenseSupport