realsenseai/librealsense #8762: Library Coordinate Guro & Accel for WF or Unity

anomal3, 2021-04-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   {Windows 10 LTSC  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.43.0}                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {others }                          | 

### Issue Description
Good afternoon, ladies and gentlemen.
I made code like this
```csharp
public static class GA
    {
        struct Vector3
        {
            public float x, y, z;
        }
       public static double[] GetDataGyro(int rounding)
        {
            var pipeline = new Pipeline();
            var cfg = new Config();
            cfg.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
            cfg.EnableStream(Stream.Color, 1280, 720, Format.Rgb8, 30);
            pipeline.Start();

            while (true)
            { 
                using (var frameset = pipeline.WaitForFrames())
                {
                    var gyroFrame = frameset.FirstOrDefault<Frame>(Stream.Gyro, Format.MotionXyz32f).DisposeWith(frameset);
                    var Gyro = Marshal.PtrToStructure<Vector3>(gyroFrame.Data);
                    return new double[] { Math.Round(Gyro.x, rounding), Math.Round(Gyro.y, rounding), Math.Round(Gyro.z, rounding) };
                }
            }
        }
}
```

Collected the library and tried it like this
```chsarp
 static void Main(string[] args)
        {
            while (true)
            {
                double[] gc = GA.GetdataShortGyro(2);
                Console.WriteLine("x{0} y{1} z{2}", gc[0], gc[1], gc[2]);
                
                GC.WaitForPendingFinalizers();
            }
        }
```
In the console application, everything works as it should. But if I use WinForms then I get the common error
```
ExternalException: rs2_pipeline_wait_for_frames(pipe:0000015545087C20)
Rethrow as Exception: Frame didn't arrive within 5000
```
in the code
 ```cs 
using (var frameset = pipeline.WaitForFrames())
```

Disconnecting the camera and reconnecting does not solve my problem.

I get exactly the same thing in Unity.

*Questions:*
**1. How to make it work in Windows Forms?
2. How to make it work in Unity?**

p.s. Although it seems to me that if this works in Windows Forms then it will work in Unity too

anomal3, 2021-04-07:
```cs
cfg.EnableStream (Stream.Gyro, Format.MotionXyz32f); 
cfg.EnableStream (Stream.Accel, Format.MotionXyz32f);

_pp = _pipeline.Start (cfg, f =>

{

var profile = f.Profile;

});
```

I don't know how to apply the callback.
I'm in stupor. Help me to understand

Why do I need PiplenProfile ? that _pp?
[image]

MartyG-RealSense, 2021-04-07:
Hi @anomal3  I wonder whether the problem is with use of the highlighted **FirstOrDefault** type of frameset command, given the complications of accessing IMU data described in the links below).

https://github.com/IntelRealSense/librealsense/issues/4803#issuecomment-528331115

https://github.com/IntelRealSense/librealsense/issues/5291#issuecomment-563436161

The **Linq Carefully** script in the SDK's **C# Cookbook** page of C# examples illustrates how **f =>** can be used.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#linq-carefully

anomal3, 2021-04-07:
> Hi @anomal3 I wonder whether the problem is with use of the highlighted **FirstOrDefault** type of frameset command, given the complications of accessing IMU data described in the links below).
> 
> [#4803 (comment)](https://github.com/IntelRealSense/librealsense/issues/4803#issuecomment-528331115)
> 
> [#5291 (comment)](https://github.com/IntelRealSense/librealsense/issues/5291#issuecomment-563436161)
> 
> The **Linq Carefully** script in the SDK's **C# Cookbook** page of C# examples illustrates how **f =>** can be used.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#linq-carefully

I followed your advice, made a callback. And now I get this
[image]

**Object does not support "librealsense::composite_frame" interface! "**

anomal3, 2021-04-07:
Ohhh  Even an example
[image]

anomal3, 2021-04-07:
> Hi @anomal3 I wonder whether the problem is with use of the highlighted **FirstOrDefault** type of frameset command, given the complications of accessing IMU data described in the links below). [...]

Why did you shut up @MartyG-RealSense?

anomal3, 2021-04-08:
I realized this is impossible

MartyG-RealSense, 2021-04-08:
I apologise for the delay in responding further due to a high number of RealSense customer support requests at this time.

anomal3, 2021-04-09:
> I apologise for the delay in responding further due to a high volume of RealSense customer support requests at this time. Do you wish to continue with this case please?

Yes of course. only I have no idea how to move on

MartyG-RealSense, 2021-04-09:
Further research of your problem found a link to a past case I handled about a Unity project for accessing the IMU in Unity with D435i.

https://github.com/IntelRealSense/librealsense/issues/5993#issuecomment-596358982

anomal3, 2021-04-12:
This is the best solution to solve my problem

closed 2021-04-12 by anomal3