realsenseai/librealsense #8766: rs-server/net with realsense-viewer error

ojjsaw, 2021-04-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 - D435i } | 
| Firmware Version                   | (05.12.12.100) | 
| Operating System & Version |   {Win (10) / Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  NA                                        | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  { 2.43.0 }                          | 
| Language                            |  { NA }                          | 
| Segment			|  {others }                          | 

### Issue Description
Trying the streaming with rs-server on raspberry pi and just realsense-viewer > add Network Device.
It seems to detect the d435i camera from the raspberry pi (AddSource shows d435i IP and S/N) but shows below error in pop-up after selecting.

Realsense error calling:
Invalid Value in rs2_get_device_info(dev:0x7700840, info:Product Id):
info Product Id not supported by the device!

MartyG-RealSense, 2021-04-07:
Hi @ojjsaw  Have you used Intel's **open-source ethernet networking** guide in the link below to set up an ethernet network for the camera before trying to access the camera through Add Source > Add Network Device please?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

ojjsaw, 2021-04-07:
@MartyG-RealSense Yup..

Update: 
- If I write the code myself basic opencv c++ application with net api the setup actually works perfectly! 
- The issue seems to be present only in "realsense-viewer" application. Which was my 1st go-to test to see if its worth proceeding this direction.

MartyG-RealSense, 2021-04-07:
There was another case reported yesterday of this particular warning box when setting the network camera in the RealSense Viewer.  The RealSense user in that case encountered the issue in SDK 2.44.0. 

https://github.com/IntelRealSense/librealsense/issues/8529#issuecomment-814275813

I will highlight these two instances of this error to Intel.  Thanks for the report!

MartyG-RealSense, 2021-04-08:
Can you confirm that you are using a stream format with a resolution and FPS that is supported by that format please?  The list of modes supported by rs-server is in a table in the link below.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

MartyG-RealSense, 2021-04-09:
Thanks very much @mirkocomparetti-synesis for sharing your findings!

3dsf, 2021-04-09:
see #8675

MartyG-RealSense, 2021-04-15:
Thanks @3dsf for your link.

@ojjsaw  Do you require further assistance with this case, please?  Thanks!

ojjsaw, 2021-04-15:
@MartyG-RealSense  Nope.. just wanted to bring to notice - rs-server/net_device API works in custom application and only in realsense-viewer app throws an error upon adding a new network device in v2.43.0. No further assistance required on this topic.

MartyG-RealSense, 2021-04-15:
Okay, thanks very much @ojjsaw for bringing it to my attention.  :)    I will close the case now as you do not require further assistance.  Thanks again!

closed 2021-04-15 by MartyG-RealSense

AndreV84, 2021-05-23:
which would be the most straightforward way to do rtp/rtsp streaming from depth camera to remote computer in the network? 
does it also work for l515 or only d400x?
I can see that we need to pass the cmake flag
```
  -DBUILD_NETWORK_DEVICE=ON to the list of CMake flags 
```
so that rs-server realsense2-net will get build,
but does it need to be done on both computers?
if so which further modificationn will be required at each side?
I can see:
```
Add #include <librealsense2-net/rs_net.hpp> to the code
Inside CMakeLists.txt add realsense2-net to target_link_libraries
Create network device from IP and add it to a context:
rs2::net_device dev("10.0.0.99");
rs2::context ctx; dev.add_to(ctx);
Pass context to pipeline constructor:
rs2::pipeline pipe(ctx);
```
but to which exactly file do we add
#include <librealsense2-net/rs_net.hpp> 
given we would just want to retrieve the stream at network computer using realsense-viewer? to which file does it go if we would try retrieving the realsense over the network using e.g. gstreamer? Does the realsense rtsp url get rendered so we could connect to it with gstreamer?
To where goes the following line?
```
Inside CMakeLists.txt add realsense2-net to target_link_libraries
```
is there an example of resulting file?
to which file do lines below go?
```
rs2::net_device dev("10.0.0.99");
rs2::context ctx; dev.add_to(ctx);
Pass context to pipeline constructor:
rs2::pipeline pipe(ctx);

mirkocomparetti-synesis, 2021-05-26:
We have the same doubts and question as @AndreV84 
Help would appreciated.

3dsf, 2021-05-26:
@AndreV84 @mirkocomparetti-synesis 

hope this helps 

- **both** machines require `-DBUILD_NETWORK_DEVICE=ON` 
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-2-building-from-source
- works with d400 series and d515 
- look at rs-server 
https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-server/RsServer.cpp 
- maybe something like this may interest you 
https://github.com/bmegli/realsense-ir-to-vaapi-h264

mirkocomparetti-synesis, 2021-05-26:
@3dsf which one of those examples is a network client for the rs server that will get depth and rgb data from the network camera? https://github.com/IntelRealSense/librealsense/tree/master/examples
Thanks!

AndreV84, 2021-05-26:
@3dsf 
you wrote : " works with d400 and d515"
does it work with l515? d455?

MartyG-RealSense, 2021-05-26:
Hi @AndreV84  If you require advice from Intel about your question, could you create a new case please to ask about L515, as this was originally a 400 Series related case.  Thanks!

3dsf, 2021-05-26:
@AndreV84 
I had rs-server/viewer working last fall with the **l515** https://github.com/IntelRealSense/librealsense/issues/8675#event-4795639767, but this seems to be in continual flux (rs-sever does not currently compile (2.45) for me.  

I have yet to use a d455, but it worked fine with the 435i and I imagine operationally they are very similar.

3dsf, 2021-05-26:
@mirkocomparetti-synesis 
I don't think there is standalone sample for c++, but it might be worth checking out how the realsense-viewer does it.

AndreV84, 2021-05-27:
@3dsf Thank you for following up!
However 12 days ago intel pointed out that l515 support is not implemented for network streaming
https://github.com/IntelRealSense/librealsense/issues/9014#issuecomment-841533559

3dsf, 2021-05-27:
I had it running last fall

Here is a video of the l515 frames coming from rs-server colliding with the frames of a usb d435i in real sense viewer.  I believe it was an issue with the viewer rather than something more severe.

Also I believe the same thing would have happened if it were 2 d435's.

That was 7 months ago.

AndreV84, 2021-05-31:
@MartyG-RealSense 
but despite the device model,
trying to build 245 on focal
I noticed that rs-server won't build
also there is some error with libjpeg-turbo
```
fatal: invalid reference: master
CMake Error at ibrealsense-2.45.0/build/libjpeg-turbo/tmp/libjpeg-turbo-gitclone.cmake:40 (message):
  Failed to checkout tag: 'master'

AndreV84, 2021-05-31:
it should be main, not master, right?

AndreV84, 2021-05-31:
also which argument to add to which exactly cmake file for rs-server example to get built? the command line cmake argument for network won't work

MartyG-RealSense, 2021-05-31:
The libjpeg-turbo change of GitHub branch name from 'master' to 'main' is discussed in the recent links below.

https://github.com/IntelRealSense/librealsense/issues/9073
https://github.com/IntelRealSense/librealsense/issues/9106

AndreV84, 2021-05-31:
@MartyG-RealSense 
which exactly file to edit from master to main?

MartyG-RealSense, 2021-05-31:
It does not refer to a file but to the name of the default GitHub branch of libjpeg-turbo. If you look on the edge of the row of a GitHub with the green 'Code' button you can see a drop-down menu of branch names, which is labelled 'main' by default on the libjpeg-turbo GitHub

https://github.com/libjpeg-turbo/libjpeg-turbo

AndreV84, 2021-05-31:
Intel should updat ethe CMakeLists files [ which exactly]
to change master ->main

AndreV84, 2021-05-31:
@MartyG-RealSense 
is it possible to build realsense-viewer on Windows in a way it would play network stream from network realsense rs-server?
only linux rs-viewer supported? for network clients?

MartyG-RealSense, 2021-05-31:
The Windows version of the Viewer has the ability to connect to a network camera too by left-clicking on the **Add Source** option at the top of the options side-panel and selecting **Add Network Device** from its menu.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-2-7-testing-the-camera

I recall a RealSense user who was using realsense2-net on Windows with C# and Unity.

https://github.com/IntelRealSense/librealsense/issues/8347

A RealSense user modified the open source ethernet system to use ROS networking protocols instead, though performance was not optimal.

https://github.com/IntelRealSense/realsense-ros/issues/1808

A RealSense team member advised recently that multiple cameras under net_device is not yet supported, if that is what you were asking in regard to access to other RealSense cameras.

https://github.com/IntelRealSense/librealsense/issues/6376#issuecomment-850719037

re-server does support RGB.  Like other stream types though, the resolutions supported on a network camera are more limited than a USB connection.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

I hope that the above answers will be helpful.

AndreV84, 2021-05-31:
thanks @MartyG-RealSense 
it seems we can use d400, also l515 rs-server on jetson so that the stream will be played remotely