realsenseai/librealsense #8770: How to get any object contour/edge by using D455?

yuanyuanhan327, 2021-04-07:


MartyG-RealSense, 2021-04-08:
Hi @yuanyuanhan327  If you are seeking to perform volumetric measurement of the contours of an object and can use Python then the SDK example program **box_dimensioner_multicam.py** may be suitable.  Despite its name, the program can work with a single 400 Series camera.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam

A restriction of this program is that the object needs to be placed on a surface with a chessboard pattern, so it would not be suitable if you need to analyze objects without a chessboard backdrop or without having to move them.

[image]

Another approach to object detection is to use a **neural network** to identify specific kinds of objects from an RGB image and classify them.  It can then place a 'bounding box' around that object on the image and calculate the position and rotation of the highlighted object to allow a robot arm to pick / drop and handle it without the need for a chessboard backdrop. 

https://www.youtube.com/watch?v=NdEA-esYW9I

Could you provide more information please about the kind of object detection and measurement that you are aiming to achieve so that I can guide you to suitable references.  Thanks!

yuanyuanhan327, 2021-04-08:
Thanks MartyG. This is really helpful. Currently I used the neural network to recognize the object and get the bounding box. I would like to ask how I can get the contour/border of the object in the bounding box? Do you have some references to use the depth image/point cloud to get the contour/border of the object?

MartyG-RealSense, 2021-04-08:
You could perhaps obtain the required details by extracting information about the pointcloud edges.  I recommend googling for **python extract pointcloud edges** for a range of reference sources about accomplishing this.

closed 2021-04-21 by MartyG-RealSense