realsenseai/librealsense #8771: [Jetson Xavier NX] "modinfo uvcvideo | grep "version:" issue

SunggooJung, 2021-04-08:
| Camera Model                        | L515| 
| Firmware Version                    | 01.05.05.00 | 
| Operating System & Version  |  Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |   4.9.140-tegra
| Platform                                   |NVIDIA Xavier NX  |

When I do `modinfo uvcvideo | grep "version:"`, the output cannot see any realsense string

`
uav1@sg:~$ modinfo uvcvideo | grep "version:"`
`version:        1.1.1`
`srcversion:     C80D7F82A52E8746DB51EE5`

Have any idea?

RealSenseSupport, 2021-05-15:
Hi,

What process did you use to setup librealsense on your Jetson?

Thanks

MartyG-RealSense, 2021-07-02:
Hi @SunggooJung  Do you require further assistance with this case, please?  The case will be closed after 7 days from the time of writing this if we do not hear from you.  Thanks!

closed 2021-07-08 by MartyG-RealSense