realsenseai/librealsense #8772: How to get an actual depth image from bag file?

Jonapple, 2021-04-08:
| Camera Model | { D455 } |
| Firmware Version | 05.12.10.00 |
| Operating System & Version | MacOS |
| Platform | Mac XCode |
| SDK Version | 2.2.42.0 |
| Language | C++  |
| Segment | vision, RGBD |

Hi, I'm using macOS, it seems I could only get the realsense-viewer by building the SDK. 

I got two IntelRealSense cameras, one is SR300 and another one is D455

I recorded a file with 15FPS on both Stereo and RGB stream, and I got a bag file.

I used the rs-convert tool to convert this to png images (./rs-convert -p) and ply files (./rs-convert -l). Then I found the following problems:

1. The total number of RGB images and depth images are not the same, the differences at 15 fps were small and sometimes there is no difference. But under higher fps the difference became huge. How can I match the RGB image and with corresponded depth image? (remove others cannot pair)

2. No matter what setting in the realsense-viewer I used for the recording, all the depth images I converted from the bag file are displayed as "jet color scheme", which visualized the distance relationship in color.

[image]

Below is an depth image I've got:
[image]

But I actuall want a depth image like this :
[image]

Because I want to convert the RGBD into a point cloud like this:
http://www.open3d.org/docs/release/tutorial/geometry/rgbd_image.html

3. The ply file I've got from the bag file cannot be properly displayed, it can only be opened on some online viewer. such as https://www.creators3d.com/online-viewer

Here is an example:
I could display it on the website above
[image]
But if I open it with Xcode (default) or use a [draw_geometries function (open3d library)](http://www.open3d.org/docs/release/tutorial/visualization/visualization.html) in python, there is nothing:
[image]
[image]

But I also have another ply file from online, it can be properly displayed by XCode or [draw_geometries function (open3d library)], as well as online viewers. Is there anything I need to preprocess with the ply file converted from the bag file?

4. For the SR300 camera, I found the zoom setting in RGB stream and color stream are different (I used the same resolution 640*480 and same fps)
[image]
If I want to make them are same, is there any setting I can modify before recording? Or I have to use extrinsic/intrinsics to make post-processing?

Currently, I found the intrinsics on the RGB stream and depth stream are different on both D455 and RS300.

I'm new to computer vision and my question may seem dumb for professionals. Any help will be appreciated.

MartyG-RealSense, 2021-04-08:
Hi @Jonapple  Your questions are not dumb at all.  :)  I have replied to them below.  

1.  If you have auto-exposure enabled, could you try disabling an option called **Auto-Exposure Priority** under the **RGB** controls in the side-panel of the Viewer please?  Doing so should force the FPS to be a constant rate.

[image]

2.  There is an alternative RealSense bag image-extraction tool called **rs_bag2image** that outputs depth frames as PNG images in black and white.

https://github.com/UnaNancyOwen/rs_bag2image

3.  I have not seen that particular problem before on the RealSense forums regarding a ply not being visibly rendered in XCode.  It has been reported outside of the RealSense forums though.  Here are a couple of example cases:

https://developer.apple.com/forums/thread/662104

https://stackoverflow.com/questions/64100276/ply-polygon-format-file-issue-in-ios-14

You can find further reports by googling for **.ply xcode**

4.  The SR300 and D455 models are different camera technologies (SR300 = Coded Light, D455 = Stereo Depth) that also come from different time eras (SR300 launched in early 2016 and D455 in 2020).  They also have very different imaging components and internal design.  So it is natural for there to be significant differences between the two models, including in regard to intrinsics.

You could try disabling the **Post-Processing** options in the RealSense Viewer by left-clicking on the blue icon beside them to toggle it to red (Off) to see whether it makes the depth and RGB images more equal.  I believe that the depth results are more related to the SR300's technology than the settings though.

MartyG-RealSense, 2021-04-15:
Hi @Jonapple  Do you require further assistance with this case, please?  Thanks!

Jonapple, 2021-04-21:
> Hi @Jonapple Do you require further assistance with this case, please? Thanks!

Problem solved, thanks! Really helpful :)

MartyG-RealSense, 2021-04-21:
You are very welcome @Jonapple - thanks for the update!   :)

closed 2021-04-21 by MartyG-RealSense