realsenseai/librealsense #8778: Miss alignment of depth and color images

kogan-alex, 2021-04-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |  Linux (Ubuntu 18.04.1)  | 
| Kernel Version (Linux Only)    | 5.4.0                                       | 
| Platform                                 | PC  |
| SDK Version                            |  2.30.0.1216                     | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
We are using a D415 camera and are getting color images and depth map.  We are using the Realsense SDK's align function to align both frames. We visualize the overlay in 3D. We noticed that sometimes the overlay is off (the frames are not laying tightly one on the other). Is this a known issue?

MartyG-RealSense, 2021-04-08:
Hi @kogan-alex  Does it look as though the color has problems keeping aligned with the depth when the camera is moving quickly or the camera is static and there is quick movement in the scene that it is observing?  If so, does the alignment correct itself when motion ceases?

kogan-alex, 2021-04-11:
Everything is static

MartyG-RealSense, 2021-04-11:
Would it be possible to provide an example image of the misalignment, please?

Also, can you provide information pease about how you are visualizing the image in 3D please?  Do you mean a 3D point cloud (color mapped to depth) or a 3D Display such as the left and right images in a virtual reality headset?

kogan-alex, 2021-04-12:
[image]
We get the depth map aligned with the color image from Realsense alignment. We plot the point cloud and color each point according to the color image.

MartyG-RealSense, 2021-04-12:
My understanding from the above description is that you are generating a **textured point cloud**.  Is that correct, please?

If the color-textured point cloud is being generated in real-time and the alignment is off sometimes and not all the time, that makes me think that there might be variations in FPS occurring so that depth and RGB no longer both have the same FPS.

The link below provides directions to discussions about depth / RGB sync and how a constant FPS rate may be enforced.

https://github.com/IntelRealSense/librealsense/issues/8726#issuecomment-812556071

In short though, a constant FPS can be enforced by:

-  Ensuring that exposure does not exceed an upper bound if using manual epxosure; or

-  If auto-exposure is being used, disable an RGB option called **Auto-Exposure Priority**.

https://github.com/IntelRealSense/librealsense/issues/1957

kogan-alex, 2021-04-12:
Thanks, you have understood the issue correctly.

Yes, I think this is a textured point cloud.

Does this really have something to do with the FPS (not the scene nor the camera is moving during image acquisition)?

Our fps is 6 for both streams with 40ms exposure time for depth and 151us for color. They are both well below 1/6=167ms.

MartyG-RealSense, 2021-04-12:
Would it be possible to test at 15 FPS and see whether you experience the same problems?  I ask this because problems are more likely to occur at 6 FPS that disappear at 15 FPS, as the frame rate may be slow enough at 6 that the system time-outs whilst waiting for new frames.

MartyG-RealSense, 2021-04-18:
Hi @kogan-alex  Do you require further assistance with this case, please?  Thanks!

kogan-alex, 2021-04-18:
Wasn't able to recreate the problem, if it will occur again I will reopen the ticket, thanks!

MartyG-RealSense, 2021-04-18:
Okay, thanks very much for the update!  I will close the case.  Feel free to re-open it or create a new case at a future date if the problem occurs again.

closed 2021-04-18 by MartyG-RealSense

kogan-alex, 2022-03-22:
Hello! I'm reopening this ticket since irregularities keep on happening. We have overlay issues in the color/depth maps (they are occasionally not aligned). Sometimes it is a slight misalignment and sometimes the depth frame is significantly older than the color.

Perhaps it might help to state that we updated the cameras to version 05.13.00.50, it solved dropped frames issues we had in one setup but on another setup, dropped frames keep on occurring, even with one camera streaming at 6fps.

MartyG-RealSense, 2022-03-22:
Hi @kogan-alex  On the setup affected by the lag, does it improve if you increase FPS from 6 to 15?

MartyG-RealSense, 2022-03-22:
Does a camera that works normally on one setup have problems on another, suggesting that the camera and firmware are okay and the problem is with the setup that the camera is being used with?

kogan-alex, 2022-03-22:
Just tested a working camera with another PC and there were dropped frames.

Any suggestions what in the PC might cause such problems?

Can dropped frames explain overlay issues and synchronized color/depth frames?

MartyG-RealSense, 2022-03-22:
If one stream in an aligned pair of depth and color streams is not performing smoothly then there can indeed be a misalignment where one is not keeping up with changes in the other.

As the camera performs correctly on another PC, it sounds as though the problem is with the PC that it has problems on and not the camera hardware or its firmware driver or USB cable.

On the computer with problems, are you using a different SDK version to the one on the computer that works correctly or do the two machines have apparently identical setups?

MartyG-RealSense, 2022-03-23:
In https://github.com/IntelRealSense/realsense-ros/issues/1850#issuecomment-926448603 a RealSense ROS user who was using Docker with an AMD Ryzen 7 computer with NVIDIA GeForce GTX 1660 graphics found that their FPS was fine if they only subscribed to the pointcloud but they had a significant frame drop if they subscribed to additional topics such as gyro and accel.

kogan-alex, 2022-03-27:
We were able to pin point the problem more accurately. This phenomena seems to happen when there is more than one camera connected and we get frames from each of them alternately. That is: if we connect to all cameras but wait for frames from only one, everything looks fine, but if alternate and wait for frames from one camera and then another, there is an overlay issue and artifacts. Perhaps we are interfacing erroneously with the cameras? We saw that there is some kind of "syncer" object, should we be using it?

MartyG-RealSense, 2022-03-27:
If your application is accessing multiple cameras and it is using wait_for_frames() then it is recommended that **poll_for_frames()** is used instead, as described by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

I have not got a reference that exactly matches what you are trying to do in Python.  There was a multicam C++ case at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 though where a RealSense user captured an image from each attached camera simultaneously by starting simultaneous **threads** and then joining the threads.

kogan-alex, 2022-03-29:
The problem seems to be solved by assigning a different "rs.align" object to each camera (before hand it was shared). Does that make sense?

MartyG-RealSense, 2022-03-29:
That approach does make sense to me, yes.