realsenseai/librealsense #8779: Frame didn't arrive within 15000 - Raspberry / Sdk 2.44.0 / T265

JoseDinis1, 2021-04-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265| 
| Firmware Version                   |0.2.0.951 | 
| Operating System & Version | Linux pi 5.10.17-v7l+ armv7l GNU/Linux  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | Raspberry pi 4 4Gb  |
| SDK Version                            | Sdk 2.44.0 }                          | 
| Language                            |  C++                          | 
| Segment			|                         | 

### Issue Description
Im using Raspberry Pi 4 4Gb, and constantly getting this error: Frame didn't arrive within 15000

I read lots of issues in here about this message, and most people i think solved with a updated sdk. 
But i just installed the sdk yesterday, so its updated

Im using the usb cable provided.
The error happens using Usb2 or the Usb3 ports
Im getting the error in my application (im just acessing the pose information), the viewer, and the pose example

In my application i run the pipeline.start(), and when crashes, i run the pipeline.stop(), and then pipeline.start() to try to re-iniciate without booting the application:, this works if i take out the  usb cable and connect again, meahwhile i get this:

RealSense error calling rs2_pipeline_start_with_config(pipe:0x17c9228, config:0x179dc88):
    No device connected

Thanks, apreciate some help about this.

JoseDinis1, 2021-04-09:
No feedback?

csr-kick, 2021-04-09:
See #6362 

There is at least one of these posts every week, where the T265 just randomly stops working on any non-x86 platform. Our investigations have shown there is an issue with the SDKs USB implementation or the USB code on the device, specifically around the interrupt packets and timing. The RealSense team has made it abundantly clear that this will not be addressed, and that there will be no further support of the T265 as Quote:
> We have moved our focus to our next generation of products...

There are ways to mitigate the stopping (like wrapping it in a watchdog) but its quite the pain, and never will be perfect. Maybe someday someone with the time and skillset will debug it for the community. The truth is that this is a fantastic sensor when it works, but is handicapped by poor testing and a short attention span turning into a $200 paperweight.

JoseDinis1, 2021-04-10:
csr-kick thanks

i was expecting something like this really, but also i was expecting to be donne by realsense team, because they silence reveals a lot, why they sell the device, and everywhere looks like the suport arm devices? i spend 3 days of work and 300€ in something that doesnt work.

Realsense team should make it this information ver noticeable and very clear, so people doesnt loose their time and money ... I see lots of people asking for this and they dont awnser anything, like i said makes very clear the company culture.

JoseDinis1, 2021-04-10:
Just a info, today i tried in a Raspberry 4 / Ubuntu Mate, sdk 2.44, and have the same problem

nixinator, 2021-04-12:
Just to tag this thread, having the same problem with rasberry pi4 and T265, come on intel, don't leave us high and dry on this raspberry pi 4 platform?

archsoong, 2021-06-15:
Same here with NVidia TX2 and T265, We brought two T265 so we can have extras to test, but now are stuck in the middle of nowhere and our budget is running low, and we will be able to deliver on time since we need to search for an alternative solution. Intel should address this matter publicily.

nixinator, 2021-06-15:
we accidently blew our T265 to pieces, so can no longer test new firmware/drivers etc etc.

Seems intel have washed their hands of this.

we had the realsense T265 working with TX2 fine.

DM me if you want a chat about getting that working for you.

archsoong, 2021-06-16:
> we accidently blew our T265 to pieces, so can no longer test new firmware/drivers etc etc.
> 
> Seems intel have washed their hands of this.
> 
> we had the realsense T265 working with TX2 fine.
> 
> DM me if you want a chat about getting that working for you.

Can you mail me `archsoong@gmail.com` ? cause I cant seem to find your email

julianx4, 2022-01-14:
Has there been any progress since on this?

There are many open issues here and I also saw a discuss on this topic in the raspberry pi forum where a raspberry engineer was involved: https://forums.raspberrypi.com/viewtopic.php?p=1855178&hilit=T265#p1855178
Unfortunately without any conclusion. 

I am using a t265 and d435 for a self driving rc car based on a raspberry pi 4. With the t265 crashing every 2 minutes it’s worthless. 

Did anyone find a work around? I couldn’t find anything. 

I tried using a pi 3, but it can’t manage both cameras at the same time it seems.

nixinator, 2022-01-14:
I feel your pain, it's a real finger pointer exercise, between intel <---> Rpi , there are a lot of subsystems and protocols at play.

So it's going to to take a dedicated effort between intel and RPi to get this to work.

However, Intel wants to sell you the latest and greatest hardware, which works.... so take your pick.

I really do class these devices as experimental and in beta test mode (we are the beta testers!).

Impressive hardware from intel none the less. Who to blame, intel hardware/software or Rpi 4 hardware/software , or something in between!!!

julianx4, 2022-01-14:
But intel doesn’t even have any new hardware to replace the t265 or do they? I didn’t find anything like the t265 even from another manufacturer…
At least I managed to make it run on the rpi 3 now, was just a matter of selecting the right resolution for the d435 that is supported by usb 2, but performance is 40% of what it was on the rpi 4…

wartih, 2023-07-14:
still having this problem on rpi 4 with the latest supported sdk(2.53)..

closed 2026-10-23 by MartyG-RealSense