realsenseai/librealsense #8780: Question: SR300 units of color camera exposure value

hernot, 2021-04-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 | 
| Firmware Version                   | Nort relevant | 
| Operating System & Version |   Linux (Ubuntu 18)/Mint 19 | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.X.X                          | 
| Language                            |  python                          | 
| Segment			|  others                           | 

### Issue Description
I bet there has already been request but i fail to find and missed it on the current datasheet of the SR300/SR305 type camera
What is the unit of the exposure value of the color camera?
For the IR depth camera the data sheet indicates that motioncontrol value unit is about 1ms which correlates with my experience.

MartyG-RealSense, 2021-04-09:
Hi @hernot  My research of your question did not find an answer in formal documentation sources such as the data sheet, so I do not believe that you missed it.  There is some useful information provided by RealSense team members on this GitHub forum though.

https://github.com/IntelRealSense/librealsense/issues/5493#issuecomment-567857540

https://github.com/IntelRealSense/librealsense/issues/5569#issuecomment-573495412

hernot, 2021-04-09:
Thank you very much i have seen the both but the first is about D435 using some windows specific binding and not python and not RS300 and  talking about logarithmic scale to be set through no librealesense native interface. Similar the second. So specific question still unanswered.

MartyG-RealSense, 2021-04-09:
I am unable to find a direct one-line documentation or forum quote for the information that you are seeking (about the units used by the SR300's RGB exposure on Linux) and though I can speculate, I would rather not base an answer on that without references to back that speculation up.

@ev-mp  Could you provide information about the above question please?

MartyG-RealSense, 2021-04-16:
@ev-mp Could you take a look at the above question of @hernot please?  Thanks!

hernot, 2021-04-21:
I think i figured already my self.
By setting gain to 0 exposure to max which is 10000 and observing an analog alarmglock with a seconds hand revolving at constant speed of about 1 revolution per section. In this setup the seconds hand appeared to jump aprox every second to the next position. Follows exposure 10000 likely is 1 second thus units of exposure must be 0.1 ms approx. Not a very precise measurement but a reasonable estimate.

MartyG-RealSense, 2021-04-22:
Excellent to hear that you made good progress @hernot - do you require further assistance with this case please?  Thanks!

hernot, 2021-04-22:
appart from some confirmation that my estimate isn't just random and that there are several parameters where device independent description would benefit from an update, eg white balancing parameter (not checked on D415 yet) is just the color temperature of the light source used to light the scene like on any other digital camera. a value of 2500 would assume that the scene is light by a warmwhite bulb 3900 would mean light by a fluorescent light or led, etc.

MartyG-RealSense, 2021-04-22:
Thanks very much @hernot for your feedback.  I have added a **Documentation** label to this case so that it can be considered by the RealSense team at a future date.  This case should be kept open in the meantime.  Thanks!

MartyG-RealSense, 2021-12-22:
Case closed due to the SR30x models being declared End Of Life (retired) by Intel.

closed 2021-12-22 by MartyG-RealSense