realsenseai/librealsense #8781: Error while starting Intel RealSense Viewer

SamirAL89, 2021-04-09:
| | |
|---|---|
|**librealsense**|2.42.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-04-09:
Hi @SamirAL89   Are you able to access other RealSense programs successfully with the camera, please?

You can test this in Windows by right-clicking on the launch icon for the RealSense Viewer on the Windows desktop and selecting the **Open file location** menu option.  This will open the **Tools** directory of the RealSense SDK folder where a set of pre-built executable versions of example programs are available.

Make sure that the camera is plugged in and then find the file named **rs-capture.exe** in this folder and double-leftclick on it to launch it.  Does the program launch successfully? 

[image]

SamirAL89, 2021-04-09:
Good afternoon,
The generated error is inconsistent. Sometime the errors occurs and relaunching the realsense-viewer does solve the problem, other times I was obliged to restart my system to be able to use it again. 
Regarding the rs-capture, it does work  fine without any issue. However I don't think that the working state of the rs-capture information will help because the viewer-works fine too. If the error occurs, I will immediately  try to test the rs-capture tools.

MartyG-RealSense, 2021-04-09:
Okay, thank you very much @SamirAL89 - please do update here when you have further test results.  Good luck!

MartyG-RealSense, 2021-04-15:
Hi @SamirAL89   Do you have an update that you can provide about your case, please?  Thanks!

closed 2021-04-21 by MartyG-RealSense