realsenseai/librealsense #8782: Ubuntu 20.04 debian/apt packages seem to be missing

jackjansen, 2021-04-09:
It seems something is wrong with the package repository, or something has changed and isn't documented (or I overlooked it). I am no longer able to install the Realsense SDK on Ubuntu 20.04.

The following set of lines in my `Dockerfile` worked last month:

```
RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE
RUN add-apt-repository -y "deb https://librealsense.intel.com/Debian/apt-repo focal main" -u
RUN apt-get -y update
RUN apt-get install -y librealsense2-dkms librealsense2-utils librealsense2-dev
```

But as of yesterday I get errors:

```
 => [10/23] RUN apt-key adv --keyserver keys.gnupg.net --recv-key F6E65AC044F831AC80A06380C8B3A55A6F3EFCDE || apt-key adv --keyserver hk  1.8s 
 => [11/23] RUN add-apt-repository -y "deb https://librealsense.intel.com/Debian/apt-repo focal main" -u                                  2.1s 
 => [12/23] RUN apt-get -y update                                                                                                         2.1s 
 => ERROR [13/23] RUN apt-get install -y librealsense2-dkms librealsense2-utils librealsense2-dev                                         1.8s 
------
 > [13/23] RUN apt-get install -y librealsense2-dkms librealsense2-utils librealsense2-dev:
#16 0.990 Reading package lists...
#16 1.316 Building dependency tree...
#16 1.387 Reading state information...
#16 1.433 E: Unable to locate package librealsense2-dkms
#16 1.433 E: Unable to locate package librealsense2-utils
#16 1.433 E: Unable to locate package librealsense2-dev
------
executor failed running [/bin/sh -c apt-get install -y librealsense2-dkms librealsense2-utils librealsense2-dev]: exit code: 100

```

MartyG-RealSense, 2021-04-09:
Hi @jackjansen   Are you using the same computer that you were successful with last month or did you move your Docker / Ubuntu 20.04 installation onto an Arm-based computing device such as Raspberry Pi or Jetson, please?

jackjansen, 2021-04-14:
I'm not 100% sure I used the same _host_ computer (I may have used an Intel Mac last time and an M1 Mac this time), but docker is running in Intel mode, so that shouldn't make a difference for the container/image...

I did update the `Dockerfile` to refer to the new `https://librealsense.intel.com` PPA urls (as can be seen from the log above).

MartyG-RealSense, 2021-04-14:
M1 Macs are not currently compatible with the librealsense SDK.

jackjansen, 2021-04-14:
Sorry, I must have been unclear. I am installing in an x86-64 Ubuntu 20.04 Docker image. The docker host should be completely unimportant. (And it is: everything worked fine last month). The _guest_ runs x86-64.

But I will retry on an intel linux host.

jackjansen, 2021-04-14:
... and it turns out @MartyG-RealSense was right:-) My Docker installation has quietly moved over to arm64 from amd64 architecture (without me noticing).

Sorry for the noise, closing.

closed 2021-04-14 by jackjansen

MartyG-RealSense, 2021-04-15:
No problem at all @jackjansen - thanks for the update!

maxiuw, 2022-11-08:
I follow this and it seems to be working https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

MartyG-RealSense, 2022-11-08:
Thanks very much @maxiuw for sharing what works for you!