realsenseai/librealsense #8786: Point cloud limitations in Unreal Engine

jessetemp, 2021-04-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version | Windows 10 x64 | 
| Kernel Version (Linux Only)    | |
| Platform                                 | PC, Unreal Engine 4.26 |
| SDK Version                            | 2.43.0.3013 | 
| Language                            | C++ | 
| Segment			| ? | 

### Issue Description
Hello,
I am just getting started using a RealSense camera with the Unreal Engine wrapper. I understand the point cloud feature is experimental, and uses a third-party library. The problem I'm running into is that the world normals of the voxels making up the point cloud are all pointing in the same direction.  

[image]  

Is this expected behavior?
Is it a limitation of the third party library or the way it was implemented in the UE wrapper?

I would like to turn the point cloud into a contiguous mesh with appropriate world normals, but I don't exactly know where to begin. Can anyone offer some insight on how to approach this problem?

Thanks!

MartyG-RealSense, 2021-04-11:
Hi @jessetemp  As there are no librealsense references available regarding this question, I researched about wrong-direction normals in **PCL**, as the UE wrapper instructions define the point cloud as being PCL-based.

The link below makes some suggestions about how the direction of normals might be corrected in a PCL point cloud, though I do not have information on how this knowledge might be adapted to the UE wrapper's point cloud implementation.

https://stackoverflow.com/questions/47428220/pcl-estimates-wrong-normal-direction-in-some-parts

There was also a non-RealSense PCL case where the normals of a point cloud were all pointing upwards in parts, and the user in that case thought that it might be related to the point cloud detail being sparse in those points (so PCL did not know how to process it correctly).

jessetemp, 2021-04-11:
Thanks for the info @MartyG-RealSense. I'm a little embarrassed to admit I didn't realize PCL was its own library. Just thought it was an odd abbreviation for point cloud, and that it was all handled in the runtime mesh. This at least gives me another avenue to dig into.

MartyG-RealSense, 2021-04-18:
Hi @jessetemp  Do you require further assistance with this case, please?  Thanks!

jessetemp, 2021-04-18:
Hi @MartyG-RealSense, I don't require further assistance. The UE wrapper has actually been pretty easy to work with. I made some progress with the world normals, but still have some work to do to get the mesh aligned with the normals. I was going to post the changes here when I'm done in case anyone else finds them useful
[image]

MartyG-RealSense, 2021-04-18:
Good to hear that your project has gone smoothly!  Please do post the results here when you have them.  I will close the case in the meantime as you do not currently need assistance.  The case will remain visible in internet searches for others to read.  Good luck!

closed 2021-04-18 by MartyG-RealSense

jessetemp, 2021-04-19:
Here are the changes I made to fix the world normals and make the point cloud a contiguous mesh. Instead of creating a square centered on each point in the cloud, I create a 4-sided polygon from four valid points in the cloud. There are some checks in place to make sure they are in frame and near each other. The normals are sort of hacked, I just take the cross product from the vectors along two edges of each polygon and use that for all four corners.

I also made the following two changes. The point cloud will now cull any points greater than DepthMax. And PclDensity will now reduce the number of points vertically as well as horizontally (previously it only worked horizontally).

There's still some cleanup to do around the edges and gaps to fill, but it's good enough for what I'm doing for now.

@MartyG-RealSense, I've never contributed to a public repo, but I wouldn't mind forking librealsense and submitting a pull request if this seems like a worthwhile contribution.

All of the changes take place in ARealSenseInspector::UpdatePointCloud(). For copy/paste simplicity, I've copied the entire function below, although much of it is unchanged.
```
void ARealSenseInspector::UpdatePointCloud()
{
    SCOPED_PROFILER;

    const size_t NumPoints = RsPoints->size();
    const size_t DensityPoints = (size_t)FMath::RoundToInt(NumPoints * FMath::Clamp(PclDensity, 0.0f, 1.0f));
    if (!DensityPoints) { PclMesh->SetVisibility(false); return; }

    const size_t Step = (size_t)FMath::RoundToInt(NumPoints / (float)DensityPoints);
    if (!Step) { PclMesh->SetVisibility(false); return; }

    const size_t RenderPoints = (NumPoints / Step);
    const size_t RenderVertices = RenderPoints * 4;
    const size_t RenderIndices = RenderPoints * 6;
    const size_t MaxBufferIndices = (MAX_BUFFER_U16 / 6) * 6;
    const size_t NumSections = (RenderIndices / MaxBufferIndices) + (RenderIndices % MaxBufferIndices ? 1 : 0);

#if 1
    REALSENSE_TRACE(TEXT(
        "PCL Id=%zu NumPoints=%zu "
        "RenderPoints=%zu RenderVertices=%zu RenderIndices=%zu NumSections=%zu "
        "Density=%.3f Step=%zu"),
        PclFramesetId, NumPoints,
        RenderPoints, RenderVertices, RenderIndices, NumSections,
        PclDensity, Step
    );
#endif

    const rs2::vertex* SrcVertices = RsPoints->get_vertices();
    const rs2::texture_coordinate* SrcTexcoords = RsPoints->get_texture_coordinates();

    FVector Position[4];
    FPointCloudVertex PointCloudVertex[4];
    const float Size = (PclVoxelSize * 0.5f);
    size_t SectionId = 0;
    size_t PointId = 0;
    size_t VertexId = 0;
    size_t IndexId = 0;
    size_t NumInvalid = 0;
    int VerticalCount = 0;

    while (PointId < NumPoints && SectionId < NumSections)
    {
        // PointId runs horizontally then vertically along NumPoints
        if (!PclMeshData.Contains(SectionId))
        {
            NAMED_PROFILER("AllocMeshSection");
            PclMeshData.Add(SectionId, MakeUnique<FMeshSection>());
            PclMeshData[SectionId]->PclVertices.SetNumUninitialized(MAX_BUFFER_U16, false);
            PclMeshData[SectionId]->PclIndices.SetNumUninitialized(MAX_BUFFER_U16, false);
        }

        if (!PclMesh->DoesSectionExist(SectionId))
        {
            NAMED_PROFILER("CreateMeshSection");
            PclMesh->CreateMeshSection(SectionId, PclMeshData[SectionId]->PclVertices, PclMeshData[SectionId]->PclIndices, false, EUpdateFrequency::Frequent);
            PclMesh->SetMaterial(SectionId, PclMaterial);
        }

        FPointCloudVertex* DstVertices = &(PclMeshData[SectionId]->PclVertices[0]);
[... 112 lines cut ...]
            NAMED_PROFILER("UpdateMeshSection");
            PclMeshData[SectionId]->PclVertices.SetNumUninitialized(VertexId, false);
            PclMeshData[SectionId]->PclIndices.SetNumUninitialized(IndexId, false);
            PclMesh->UpdateMeshSection(SectionId, PclMeshData[SectionId]->PclVertices, PclMeshData[SectionId]->PclIndices); // TODO: SLOW AS HELL
            PclMesh->SetMeshSectionVisible(SectionId, true);

            SectionId++;
        }
    }

    const auto NumMeshSections = PclMesh->GetNumSections();
    if (NumMeshSections > SectionId)
    {
        for (size_t i = SectionId; i < NumMeshSections; ++i)
        {
            //REALSENSE_TRACE(TEXT("hide id=%zu"), i);
            PclMesh->SetMeshSectionVisible(i, false);
        }
    }

    //REALSENSE_TRACE(TEXT("total=%zu invalid=%zu"), PointId, NumInvalid);
    PclMesh->SetVisibility(SectionId != 0);
}
```
[image]

MartyG-RealSense, 2021-04-19:
Thanks so much @jessetemp for sharing your solution with the RealSense community!

A pull-request contribution is certainly very welcome if you are able to make one.

MatthewLKomar, 2021-08-15:
@jessetemp What settings were you running your code under? I couldn't get it to work on my end with everything having a resolution of (Width=848,Height=480,Rate=30).

jessetemp, 2021-08-15:
Hi @MatthewLKomar, I set the depth and color to 1280x720 @ 30 fps and disabled infrared. It may work with other resolutions but as you've seen, some will break. I ended up reworking UpdatePointCloud a few times since posting the code here. The main problem in it's current state (aside from various bugs) is that the voxels aren't sharing vertices. They're all discrete which prevents Unreal from smoothing the edges. I haven't looked at it in a few months, but it is on my list of projects to revisit.