realsenseai/librealsense #8793: Will aligned depth image have any difference from the unaligned depth image?

Andyshen555, 2021-04-12:
As topic, this is not a bug report. I would like to know if we want to measure the distance between two points in the depth image through projection, is there any problem using the aligned image? The resolutions are the same.

MartyG-RealSense, 2021-04-12:
Hi @Andyshen555  Using alignment would not pose a problem.

The C++ SDK example program **rs-measure** demonstrates measuring between two points on an image where color is aligned to depth.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

Whilst a Python conversion of rs-measure by RealSense user **soarwing52** takes the approach of aligning depth to color.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

Andyshen555, 2021-04-12:
I just want to make sure, because we are using the cameras with ROS and the aligned depth image is recorded directly. I understand the function uses camera intrinsic parameters to perform the measurement, but the depth image is either zoomed in (D435i) or zoomed out (D455) to match the color image. The intrinsic parameters doesn't seem to change?

MartyG-RealSense, 2021-04-12:
Thanks for the information.  Are you using **align_depth = true** to publish additional ROS topics that are aligned to depth?

Andyshen555, 2021-04-12:
Yes.

MartyG-RealSense, 2021-04-12:
I carefully considered your question about ROS.  I believe that a key factor in how accurately the depth and RGB align will be the camera having a healthy **calibration**.  You can perform a quick calibration of the camera and receive a 'health check' score for the calibration in the RealSense Viewer by going to the **More** option near the top of the options side-panel and selecting the **On-Chip Calibration** menu option.

[image]

After the calibration has completed and a health-check score for the calibration has been provided, you have the option of storing the calibration in the camera or dismissing it.

[image]

Andyshen555, 2021-04-12:
Through our experiments, we noticed a constant under estimate of size from D455 and a constant over estimation from D435i. However, the camera parameter output are unchanged for depth and aligned depth image. Thus, the answer should be: don't do measurements with aligned depth in ROS.

closed 2021-04-12 by Andyshen555