realsenseai/librealsense #8803: Wrong color imagery in manual exposure mode when switching between auto-exposure and manual exposure

bulveops, 2021-04-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415/D435 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8.0-48-generic                               | 
| Platform                                 | PC|
| SDK Version                            | 2.43.0.3018 | 
| Language                            |  python                          | 
| Segment			|  Robot| 

### Issue Description
Switching between manual exposure mode to auto-exposure and back to manual exposure results in different brightness for same exposure parameters in manual exposure mode. Code to reproduce the issue:

```
import time

import pyrealsense2 as rs
import numpy as np
import cv2

EXPOSURE_TIME = 50 # 5ms
GAIN = 128

def main():
    pipeline = rs.pipeline()
    config = rs.config()

    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    #Manual exposure - first run
    profile = pipeline.start(config)
    frames = pipeline.wait_for_frames()
    color_sensor = profile.get_device().first_color_sensor()
    color_sensor.set_option(rs.option.exposure, EXPOSURE_TIME)
    color_sensor.set_option(rs.option.gain, GAIN)

    for i in range(30):
        color_frame_manual_exposure_1 = np.asanyarray(pipeline.wait_for_frames().get_color_frame().get_data())
    pipeline.stop()

    #Auto-exposure
    profile = pipeline.start(config)
    frames = pipeline.wait_for_frames()
    color_sensor = profile.get_device().first_color_sensor()
    color_sensor.set_option(rs.option.enable_auto_exposure, 1)

    for i in range(30):
        color_frame_autoexposure = np.asanyarray(pipeline.wait_for_frames().get_color_frame().get_data())
    pipeline.stop()

    # Manual exposure - second run
    profile = pipeline.start(config)
    frames = pipeline.wait_for_frames()
    color_sensor = profile.get_device().first_color_sensor()
    color_sensor.set_option(rs.option.exposure, EXPOSURE_TIME)
    color_sensor.set_option(rs.option.gain, GAIN)

    for i in range(30):
        color_frame_manual_exposure_2 = np.asanyarray(pipeline.wait_for_frames().get_color_frame().get_data())
    pipeline.stop()

    combined_image = np.hstack((color_frame_manual_exposure_1, color_frame_autoexposure, color_frame_manual_exposure_2))
    cv2.imwrite("manual_auto_manual.jpg", combined_image)
    cv2.imshow("Manual, auto, manual", combined_image)

    cv2.waitKey(30000)

if __name__ == "__main__":
    main()

```

This sample of code takes sequentially 3 frames: in manual exposure mode, then in auto-exposure mode, and again in manual exposure mode. Exposure parameters if both attempts of manual exposure mode are equal. Expected to have the same level of brightness on the captured frames in manual exposure mode. But the last frame looks overexposured.
[image]

MartyG-RealSense, 2021-04-12:
Hi @bulveops   When auto-exposure is active, the manual exposure control is not updated.  So the real exposure value has to be retrieved from **metadata**.  The links below are useful references about this subject

https://github.com/IntelRealSense/librealsense/issues/1624#issuecomment-389085387
https://github.com/IntelRealSense/librealsense/issues/3717

bulveops, 2021-04-12:
Hello, @MartyG-RealSense ,
My concern isn't about getting the actual value of exposure parameters in auto-exposure mode. I expect to have the same level of brightness in manual exposure mode if both `rs.option.exposure` and `rs.option.gain` are unchanged for every attempt. But the problem is that if enable autoexposure mode between manual exposure sessions I have different brightness of frames.
In the presented sample of code I skip 30 frames to avoid collecting frames with unsettled exposure.

MartyG-RealSense, 2021-04-12:
Auto-exposure should automatically disable when the exposure value is manually changed.  However, does it make a difference if you use the instruction **color_sensor.set_option(rs.option.enable_auto_exposure, 0)** in the second manual section to disable auto-exposure in the line directly before you set manual exposure?

bulveops, 2021-04-13:
Unfortunately there is no difference when explicitly disabling auto-exposure before manual setting of exposure option.

MartyG-RealSense, 2021-04-13:
Could you test what the result would be if the processes were re-ordered into **autoexposure, manual exposure, autoexposure** instead of the current manual, auto, manual?

bulveops, 2021-04-14:
The sequence **autoexposure, manual exposure, autoexposure, manual exposure**. The last manual exposure section results in an overexposed frame.
[image]

bulveops, 2021-04-14:
Interestingly that the sequence of code sections **manual exposure, autoexposure, manual exposure, manual exposure** with operations of stopping and starting pipeline between sections allows to get correctly exposed frame for the last manual exposure section.
[image]

MartyG-RealSense, 2021-04-14:
May I ask what the purpose is in this project of switching exposure modes from manual to auto and back to manual, please?

bulveops, 2021-04-14:
Sometimes it is reasonable to delegate the responsibility of choosing exposure mode to the user. It is supposed that switching between different modes in the same session doesn't result in broken imagery, I think.

MartyG-RealSense, 2021-04-15:
It should not result in broken imagery, no, and some RealSense users have chosen to toggle between auto and manual exposure during the same session (for example, to set up the initial exposure values for a scene with auto-exposure and then switch to manual exposure). 

It may be useful to print the exposure value for each stage to confirm that on the frames that are over-exposed, they are set to the manual exposure value that they are meant to be.

bulveops, 2021-04-16:
The actual exposure and gain values obtained with `color_sensor.get_option(rs.option.exposure)` `color_sensor.get_option(rs.option.gain)`  are exactly the same I set manually.

MartyG-RealSense, 2021-04-16:
Given that manual exposure does not need time to settle down like auto-exposure does (and so does not require the 'skip several frames' mechanism, the next action that you could try to see whether it makes a difference is removing the 'for i = 30' instruction from the manual exposure sections and only using it for the auto-exposure section.

bulveops, 2021-04-16:
Unfortunately no difference

MartyG-RealSense, 2021-04-16:
Rather than stopping and starting the pipeline, would it be possible to have the entire process run in a single pipeline session from start to finish, with the pipeline closed after the final frame of the sequence is captured?

[image]

bulveops, 2021-04-16:
In the entire process run in a single pipeline session there is no problems with overexposure. What goes wrong if restarting a pipeline session between exposure modes?

MartyG-RealSense, 2021-04-16:
The sense that I got was that it may not be ideal to keep repeating the same configuration instructions each time when some of the details could perhaps be defined only on the default pipeline start when the program launches.  An example of this would be **frames = pipeline.wait_for_frames()** and **color_sensor = profile.get_device().first_color_sensor()** ... my thinking is that frames and color_sensor should still have the same definitions in subsequent starts once their configuration is defined in the initial start.

MartyG-RealSense, 2021-04-22:
Hi @bulveops  Do you have an update that you can provide about this case, please?  Thanks!

bulveops, 2021-04-22:
Hi @MartyG-RealSense. Probably it could be a solution in an application. But it seems that different behaviour when restarting session and changing exposure modes is a kind of bug. BTW thank you for your assistance!

MartyG-RealSense, 2021-04-22:
I will leave this case open for a further time period to give you the opportunity to perform further tests if you wish to and give feedback here  Otherwise, feel free to close the case if you are satisfied.  Thanks!

MartyG-RealSense, 2021-05-04:
Hi @bulveops   Do you wish to continue with this case, please?  Thanks!

closed 2021-05-10 by MartyG-RealSense