realsenseai/librealsense #8806: Real Time Point Cloud Visualization

karlita101, 2021-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hi there, I noticed that the [https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb] by @dorodnic isn't available anymore ( nothing is loading). Will this be available again soon? 

Are there any other recommended Point cloud visualizations that  are ideal for real time visualization and that do not require PLY exporting/loading ?

[image]

MartyG-RealSense, 2021-04-12:
Hi @karlita101  Thanks very much for the report.  I tested all of the Jupyter examples and you are correct.  None of them are rendering, though the source code of the pages is still accessible through the **Display the source blob** option highlighted below.

[image]

I have passed on your report to Intel.  Thanks again!

In regard to alternative real-time point cloud examples for Python, the SDK example program **opencv_pointcloud_viewer.py** may meet your needs.

https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/opencv_pointcloud_viewer.py

Another approach is to make use of **Open3D**, like in the script in the link below:

https://github.com/intel-isl/Open3D/issues/473#issuecomment-508139946

The **pyntcloud** point cloud library, which is compatible with 400 Series cameras, can also be recommended.

https://github.com/daavoo/pyntcloud

MartyG-RealSense, 2021-04-13:
Hi @karlita101   The Jupyter Notebook tutorials are accessible again now, including the pointcloud one that you needed.

https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb

karlita101, 2021-04-17:
Great, thank you so much!

MartyG-RealSense, 2021-04-18:
If you believe that the color stream is causing your program to lag, you could try putting depth and color on separate pipelines.  An example of doing so in Python is provided below, in a script with two pipelines that puts IMU on one stream and depth / color on the other.

https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

karlita101, 2021-04-18:
> If you believe that the color stream is causing your program to lag, you could try putting depth and color on separate pipelines. An example of a multi-pipeline application in Python is provided below, in a script with two pipelines that puts IMU on one stream and depth / color on the other.
> 
> [#5628 (comment)](https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238)

Thank you Marty I will check it out. I added a few checkpoints and indeed saw that the alignment process was talk ~50% of the processing time, the other half from updating the visualizer.

**I have tried the **opencv_pointcloud_viewer.py** and getting ~ 20 fps.** 

For my application, I am looking to only generate a point cloud for a specific area of interest (ROI) defined by a the bounding area of aruco markers detected by in the RGB frame. So I implemented the rs.align () process as in my code above and in the rs. align example. 

Here I notice that the alignment process does not take nearly as much time as in the open3D script. It's still not quite clear to me how the align process would be much quicker with this implementation if it follows the same code ( ~0.008s). **1)Would you have any insight into this?**

`**align_to = rs.stream.color
align = rs.align(align_to)**

while True:
    # Grab camera data
    if not state.paused:
        dt0 = datetime.now()
        
        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        
        # depth_frame = frames.get_depth_frame()
        # color_frame = frames.get_color_frame()
        
        #KR
        **#Align the depth frame to color frame
        aligned_frames = align.process(frames)

        #Get aligned frames        
        depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame(**)`

**2) The SDK method initializes the entire point cloud object before the depth frame is aligned to the colour frame. Would this cause any issues?

3) Is it possible to change the depth_frame to the aligned_depth_frame, and ONLY specify for a specific ROI? 
-I already have a binary mask for the ROI, so I wanted to apply a cv2.bitwise operation to get only the points where the mask is true... which why I was trying to use open3d since I can work directly with the depth_image and not the frame.
- It just isn't clear to me if you can pass an image into pc.calculate or only specify specific pixel ranges of an entire.**

-  points = pc.calculate(depth_frame)==> pc. calculate(aligned_depth_frame[ specify range here])

Thank you

MartyG-RealSense, 2021-04-18:
1.  In general, depth to color alignment is a processing-intensive activity.  You can take advantage of **CUDA** acceleration of depth to color alignment if you have a computer / computing device with an Nvidia graphics GPU (which includes Nvidia Jetson boards) and build librealsense with CUDA support enabled.

2.  Other than the processing demands of alignment mentioned above that may slow down the program on low-power hardware, I cannot think of anything to warn about.

3.  This link might provide some guidance in regard to defining a specific ROI on an aligned image with Python:

https://github.com/IntelRealSense/librealsense/issues/2769

In regard to the aligned_depth_frame aspect though, you may find useful the Python script in the link below for generating a point cloud with pc.calculate and aligned_depth_frame.

https://github.com/IntelRealSense/librealsense/issues/2526 

If you are primarily concerned with obtaining the details of a single specific RGB pixel coordinate instead of generating an entire aligned cloud, then an alternative that should have faster processing is to use **rs2_project_color_pixel_to_depth_pixel**

https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-614261704

karlita101, 2021-04-20:
Thanks Marty, will look into all the suggestions!

closed 2021-04-20 by karlita101

sanjaiiv04, 2023-11-15:
@MartyG-RealSense Hey When I run the **opencv_pointcloud_viewer.py** , I realized that there is some kind of a projection of objects in the point cloud. This is the result of running the script: 
[image]
You can see a projection of my body behind me. Is this supposed to happen? How do you take care of it? I am new to point clouds and this is a little confusing to me.

sanjaiiv04, 2023-11-16:
@MartyG-RealSense I think its how the camera works. Plane surfaces like floors and ceilings are being captured and rendered properly as point clouds but other objects tend to be having the offset shadow behind them. I guess this is due to the fact that RealSense has LIDAR for depth perception and as light cannot travel through objects, there is an offset of shadow behind them. What you have sent as a reference is a flat plane and hence it renders perfectly.

MartyG-RealSense, 2023-11-16:
Thanks @sanjaiiv04   The L515 has some qualities in its depth analysis that react differently from the 400 Series cameras when observing certain types of surface.  For example, if an L515 tries to depth sense a transparent plastic bottle of water then it sees through the bottle and renders the objects behind the bottle on the depth image.

Setting the L515 visual preset **Short Range** can help to deal with this kind of situation.  Below is an example of Python code for doing so.

```
option = rs.option.visual_preset
preset = rs.l500_visual_preset.short_range
depth_sensor = depth_sensor.set_option(option, preset)
```