realsenseai/librealsense #8808: Align between RGB and Point Cloud using only PLY

nikzadb, 2021-04-12:
**Issue Description**

I have been provided with a PLY file from Pyrealsense2. I don't have access to the software or the camera and my only source of information is the PLY file. The PLY file has (x, y, z, r, g, b) format which the first three are values between -0.5, 0.5.

Now, I would like to create a 4D image from this PLY that matches the corresponding 2D image to its point cloud value; it seems  there are functions like "align" which all require things like:
```
frames = pipeline.wait_for_frames()
```
in the beginning of the code. I am wondering how can I call this line when I only have PLY code and no access to the hardware/software?

MartyG-RealSense, 2021-04-12:
Hi @nikzadb   If this is a one-off project then instead of creating an application to process the ply, it may be worth first trying to process the ply file in the open-source **MeshLab** software.

https://www.meshlab.net/

The link below has a guide to loading a RealSense-created ply into MeshLab:

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

nikzadb, 2021-04-12:
Thanks @MartyG-RealSense 
Great tool. 
However, I need to create a Python code as the application is ongoing and will be part of a bigger system

MartyG-RealSense, 2021-04-12:
I researched your question further.  It sounds as though your goal is to convert 3D ply information to a 2D **color map**.  Is this a correct interpretation, please?

If it is, the commonly recommended approach seems to be to perform the conversion using **scalar fields**.  

An example of this is described in the documentation of the **pyntcloud** Python point cloud library.

https://pyntcloud.readthedocs.io/en/latest/scalar_fields.html?highlight=color%20map#scalar-fields

And a reference from **CloudCompare**, another point cloud library:

http://www.danielgm.net/cc/forum/viewtopic.php?p=15468&sid=09f5e75a63bf0876ccff1d87ad1c5475#p15468

I hope that this provides a useful research lead.

nikzadb, 2021-04-12:
that's great @MartyG-RealSense 
i'll have a look

closed 2021-04-13 by nikzadb