realsenseai/librealsense #8810: D455 distortion

ste93, 2021-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.8.0-48  | 
| Platform                                 | PC |
| SDK Version                            |  2.0 | 
| Language                            |  c++  | 
| Segment			| robot  | 

### Issue Description
Hi all, 

I'm using a D455 with ros and yarp and I found that the distortion returned from the camera has parameters different from 0 while the d435 has paraameters equal to 0. I was wondering if the camera returns directly the image undistorted (and so we can ignore the parameters) or the image is distorted and I have to correct it manually, since from the image i cannot see whether is undistorted or not.

Thanks

MartyG-RealSense, 2021-04-12:
Hi @ste93  Are you referring to the coefficients for the **color** stream of the D455 please?  If so then yes, it has been observed that the D455 has non-zero coefficients for the color profile, whilst in general on the 400 Series cameras the profiles have their coefficients all set to zero deliberately by design.

https://github.com/IntelRealSense/librealsense/issues/8325

As a general rule, most streams produced by the 400 Series cameras are rectified except for the **Y16** format, which is unrectified as it is used for camera calibration.

ste93, 2021-04-12:
Hi @MartyG-RealSense, yes I am referring to the color stream. Should I treat it as undistorted although the parameters are different from zero or I have to undistort it? I know the streams generated by the 400 are rectified but I was wondering if the d455 is an exception or it follows that rule and it is a bug the parameters different from 0.

MartyG-RealSense, 2021-04-12:
I do not have information regarding why the color coefficients on D455 are non-zero.  However, **rs-enumerate-devices -c** (which lists the calibration information for all the camera profiles) states that the color streams of D455 have the **Inverse Brown-Conrady** distortion model.

[image]

ste93, 2021-04-12:
Thank you, but I know the distortion models. The only doubt that I have is if the camera already sends undistorted images or I have to undistort them, but I think the only method to find it is to use a checkboard since nowhere is written in realsense documentation,

agrunnet, 2021-04-12:
I can confirm that D455 third color sensor is NOT undistorted. It is sent is distorted. By contrast the depth image and left and right images have been rectified and undistorted before sending.

ste93, 2021-04-13:
thank you @agrunnet

MartyG-RealSense, 2021-04-19:
Hi @ste93  Do you require further assistance with this case, please?  Thanks!

ste93, 2021-04-19:
Hi @MartyG-RealSense , no for now it is ok, I'm waiting for the fixes. Thanks!

MartyG-RealSense, 2021-04-19:
Thanks very much for the update @ste93 - I will close this case as you do not currently require further assistance.

closed 2021-04-19 by MartyG-RealSense