realsenseai/librealsense #8812: No frames received at 1280x720x6fps on Network Device

sachin-pri, 2021-04-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 IP Device | 
| Firmware Version                   | 2.42.0 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC, Raspberry Pi  |
| SDK Version                            |   2.42.0                         | 

### Issue Description

I have been following the tutorial: https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras and have been able to view the depth and rgb images in realsense-viewer. I have no issues when streaming the data at 640x480x30fps (Depth Z16, Color YUYV). However, at 1280x720x6fps (Depth Z16, Color YUYV), I only see the color data, and I get a "No Frames Received" message in the depth image window.

[image]

MartyG-RealSense, 2021-04-13:
Hi @sachin-pri   1280x720 at 6 FPS is certainly supported for both depth and color on an rs-server network camera connection.  

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

I note that you were successful with streaming both depth and color at 640x480 at 30 FPS.  Have you tested please whether color or depth can stream successfully at 1280x720 at 6 FPS on their own and not at the same time please?  If one stream on its own can run successfully at that resolution then it would suggest a situation where two simultaneous streams at this resolution is over-burdening the network connection.

sachin-pri, 2021-04-13:
Hi @MartyG-RealSense, thank you for your response. The result when using individual streams is the same as when running simultaneously; the color comes in at 1280x720 at 6fps with no issues, and the depth does not receive any frames.

MartyG-RealSense, 2021-04-14:
Is the Raspberry Pi that you are using a Pi 4 or a lower-configuration model such as Pi 3b, please?

sachin-pri, 2021-04-14:
It is a Raspberry Pi 4

MartyG-RealSense, 2021-04-15:
Does disabling auto-exposure have any effect?

sachin-pri, 2021-04-19:
Hi @MartyG-RealSense, I don't plan to work on this for another few weeks. I'll give this a shot then and add a comment if I have an update. Thanks!

MartyG-RealSense, 2021-04-20:
Okay, thanks very much @sachin-pri for the update!  I will keep this case open for a further time period.

closed 2021-06-20 by MartyG-RealSense