realsenseai/librealsense #8813: intel realsense 3d Fail after showing 3d image for less than a minute in realsenseviewer. while 2D still works pretty well.

asterC97, 2021-04-13:
| | |
|---|---|
|**librealsense**|2.44.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|017322072137|
|**Firmware Version**|05.12.12.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|017523021869|
|**Firmware Update Id**|017523021869|

Please provide a description of the problem

asterC97, 2021-04-13:
this happends after i install the latest firmware on the depth sensor. But if i use virtualbox ubuntu realsense viewer. i am still able to reliably obtain data from realsense 3d

MartyG-RealSense, 2021-04-13:
Hi @asterC97  Could you try installing on Windows the previous **2.43.0** SDK using the **Intel.RealSense.SDK-WIN10-2.43.0.3013.exe** installer file in the **Assets** file list and see whether 3D performance is more stable, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0

[image]

asterC97, 2021-04-13:
I have installed that sdk file. its only the window version of realsense viewer that has that kind of issue. I tested in my virtualbox in ubuntu. no issue. I also tried using external software like opencv on windows to access 3d image, no issue with that. It occur recently when i tried to update my camera firmware. The window version for realsense viewer started with normally, but after a minute or so, the 3d image will begin to rapidly flicker till it blank out

asterC97, 2021-04-13:
I have followed according to what you have stated. but the bug still persist.

MartyG-RealSense, 2021-04-13:
And depth streaming worked normally in the Viewer until you upgraded to firmware **5.12.12.100**?

asterC97, 2021-04-13:
The Viewer works fine before i upgrade the firmware to 5.12.12.100. but the main issue is, if I use VirtualBox Ubuntu and use the Realsense viewer from ubuntu. I am able to stream the video without issue. I even tried using OpenCV on window 10 to obtain 3D data, and i am able to receive and process the data accordingly. The only thing i cant do is using Realsense Viewer 3D view for a long duration, when the Realsense Viewer turn on for the first time, i am able to stream for a minute before it begins to blink violently and turn blank which the 3D viewer stop streaming data

MartyG-RealSense, 2021-04-13:
Could you try disabling the two **GLSL** options in the Viewer's **Settings** interface please to see whether it makes a positive difference to the 3D point cloud rendering?  You can access this interface with the following steps.

**STEP 1**
Left-click on the gear-wheel in the top corner of the Viewer and select **Settings** from its menu to open the Settings overlay.

[image]

**STEP 2**
Left-click on the **Performance** tab on top of the Settings window and left-click on the blue boxes beside the two **GLSL** options to turn their boxes black (Off).

[image]

When done, left-click on the **Apply** button at the bottom of the Settings overlay.

**STEP 3**
Enable the depth stream.

asterC97, 2021-04-14:
After following your step, it works nicely.

MartyG-RealSense, 2021-04-14:
Great to hear, @asterC97 - you are very welcome, and thanks for the update!

MartyG-RealSense, 2021-04-16:
Hi again @asterC97   After discussion with Intel, we are re-opening this case so that we can file a bug-fix report to look at the problem of the point cloud image disappearing for a small number of RealSense users when GLSL is enabled.  

This case should be kept open whilst the bug-fix investigation is active.  You do not need to do anything further.  Thanks again!

MartyG-RealSense, 2026-10-29:
Case closed because problems with the point cloud disappearing when GLSL is enabled have not been reported for years.

closed 2026-10-29 by MartyG-RealSense