realsenseai/librealsense #8814: Align between camera and depth images

nikzadb, 2021-04-13:
I have three files:
- Camera image
- Depth image
- PLY file 
and don't have access to the actual hardware. Trying to find a way to align Camera and Depth images with what I already have. There is a code in Python that requires to get the frames from the device and do this alignment:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

however, this is not going to work for my case due to a lack of access to the device. Is there any way to do offline alignment?

Thanks

MartyG-RealSense, 2021-04-13:
Hi @nikzadb   Is this case a continuation of case https://github.com/IntelRealSense/librealsense/issues/8808 that you closed, please?  Thanks!

nikzadb, 2021-04-13:
@MartyG-RealSense: it is connected but not the same issue.
the previous one is mainly about PLY data but this one is mainly about alignment between Camera and Depth images.

Back to the issue:
- is there any connection between RGB in PLY data and Camera image? It looks not.
- "align" function is the main way to do this alignment; however, I don't have access to the main Hardware. It is possible (and how) to run "align" function on Depth/Camera images + PLY?

MartyG-RealSense, 2021-04-14:
If it were a **bag** format file where the depth and color data is stored individually in the file then such an alignment operation should be straightforward.  This is because once a bag-file recording is loaded into memory then further processing such as alignment and post-processing can be performed on the individual streams in real-time.

Typically, if a point cloud is exported to ply with color information included then the color has already been mapped to the depth points.  This process of exporting from the SDK a **textured point cloud** as a ply is described in the guide in the link below.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

So I would think that if RGB was mapped to depth at the time that the ply was created then carrying out a depth to color alignment operation on the ply-file data would be unnecessary.  If you do not have access to the camera hardware then it is not possible to do this though, of course.

In regard to an alignment solution for ply and PNG images, I'd speculate that it might be possible if you converted the ply back to a point cloud and then mapped the PNG image to the generated point cloud's coordinates, similar to the real-time mapping of RGB onto depth in the 3D mode of the RealSense Viewer.

Perhaps the ply to point cloud conversion could be done in PCL using a feature such as **PLYReader**:

https://pointclouds.org/documentation/classpcl_1_1_p_l_y_reader.html

nikzadb, 2021-04-14:
Thanks @MartyG-RealSense 
i used PyntCloud to read the point cloud. As expected, there are 6 values (x,yz,r,g,b):
- my understanding is that "z" is the distance from the sensor. is it correct to multiply "z" with depth scale (0.25 for L515) to gatactual distance in meter?
- from the PLY data and its rgb, is it possible to reconstruct the Camera image?

MartyG-RealSense, 2021-04-15:
1.  Yes, the real-world distance in meters can be calculated by multiplying the 16-bit pixel depth value by the depth scale.   For the 400 Series cameras, the default depth scale is **0.001**.  So if your application does not change the depth scale then it can be easier just to multiply by a fixed value of 0.001 than read the depth scale value in real-time with programming instructions.

I cannot comment on L515 values as I do not currently handle L515-related questions.  You can check the depth scale for a particular camera model though by opening the RealSense Viewer with that camera attached and looking at the 'Depth Units' setting under the **Controls** category of the Stereo Module options.

2.  When you say the **Camera** image, do you mean converting a textured point cloud back to a 2D RGB image, please?

MartyG-RealSense, 2021-04-21:
Hi @nikzadb  Do you require further assistance with this case, please?  Thanks!

nikzadb, 2021-04-21:
Apperatly the only way is to use align function. 
after using this on the hardware the RGB image and depth lidar data align well

closed 2021-04-21 by nikzadb