realsenseai/librealsense #8815: the pose of apriltag is not right (D455 1280x720),

ctxqlxs, 2021-04-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 

### Issue Description
the pose of apriltag is not right (D455 1280x720),  do we need to do something like [this]?(https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/examples/pose-apriltag/rs-pose-apriltag.cpp#L121)

MartyG-RealSense, 2021-04-13:
Hi @ctxqlxs  The SDK's pose examples are designed for use only with the RealSense T265 Tracking Camera.  The 400 Series stereo depth cameras do not have support for pose data built in.  This is mentioned in the introductory lines of the documentation of the pose examples.

https://github.com/IntelRealSense/librealsense/tree/master/examples/pose-apriltag#rs-pose-apriltag-sample
https://github.com/IntelRealSense/librealsense/tree/master/examples/pose#rs-pose-sample

It is possible to use a different type of fiducial image marker called **ArUco** with the 400 Series cameras though.

An article on the official RealSense blog about the use of the D435 camera model with **Everdrone** medical flying drones shows how Everdrone makes use of marker images to guide the landing of the drone.

https://www.intelrealsense.com/everdrone/

https://www.youtube.com/watch?v=e_aGS1pEDIw

[image]

An example of a 2DOF ArUco pose estimator in ROS that was tested with D435 is in the link below.

https://github.com/Jphartogi/ipa_marker_detection

ctxqlxs, 2021-04-13:
You've misunderstood. @MartyG-RealSense 
I didn't use the code referenced above. 
640x480, it seems right:
[image]
but 1280x720:
[image]
and i have changed the camera intrinsic.

MartyG-RealSense, 2021-04-13:
What program are you using for the tag detection please?

Normally the distortion coefficients are all set to zero automatically o the 400 Series cameras.  On the D455 model, the coefficients of the color profiles are an exception to this and report as non-zero.  This subject was discussed on another case yesterday.

https://github.com/IntelRealSense/librealsense/issues/8810

closed 2021-04-14 by ctxqlxs