realsenseai/librealsense #8818: Hand Tracking by intel realsense D415

L-xn, 2021-04-13:
I am using D415 for hand tracking. I found that there is no bone recognition content in SDK 2.0. I sincerely ask the experts how to use D415 to recognize the 22 joints of the hand and how to recognize the joints import into Unity. There is very little information about D415 hand tracking. Where can I find more information and procedures about this. Hope everyone help me.Thanks!
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 | PC  |
| SDK Version                            |  { intel realsense SDK2.0 }                          | 
| Language                            |  {C++/python/unity }                          | 
| Segment			|  {Hand tracking }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-04-13:
Hi @L-xn   You are correct that RealSense SDK 20 does not have built-in support for joint tracking, and I am not aware of a pre-made solution for your intended application.  It may be technically possible to develop such an application yourself though.

Intel published an advanced-level RealSense seminar on YouTube on the subject of how to use a camera to train your own data-set of body joints.  At the end of the seminar, Intel demonstrate how they were able to apply that data to the bone joints of a humanoid character model in Unity to move its body parts.

https://www.youtube.com/watch?v=VSHDyUXSNqY

L-xn, 2021-04-13:
> @ L-xn，您好 您是正确的，RealSense SDK 20不具有对联合跟踪的内置支持，并且我不知道针对您的预期应用程序的预制解决方案。但是，从技术上来说，自己开发这种应用程序可能是可行的。
> 
> 英特尔在YouTube上发布了高级实感研讨会，主题是如何使用相机训练自己的身体关节数据集。在研讨会结束时，英特尔展示了他们如何将这些数据应用于Unity中人形角色模型的骨骼关节，以移动其身体部位。
> 
> https://www.youtube.com/watch?v=VSHDyUXSNqY

Ok，thank you! Recently, I found a complete example of using realsense SR300 for hand tracking. The SDK is 2016R3. It seems I have to keep looking!

MartyG-RealSense, 2021-04-19:
Hi @L-xn  Do you require further assistance with this case, please?  Thanks!

L-xn, 2021-04-19:
> 嗨@ L-xn 在这种情况下，您需要其他帮助吗？谢谢！

I changed the device. I will use SR300 to complete my graduation project. I am studying how to build a finger motion recognition system and measure the frequency and angle of finger motion. Hope to get advice from seniors! I will be extremely grateful!

MartyG-RealSense, 2021-04-19:
You mentioned earlier the old **2016 R3** SDK for Windows.  Are you planning to use that SDK and its hand-tracking / finger-joint tracking support for your project please?

https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html

Here is a YouTube video of a tech demo I made with an SR300, 2016 R2 and the menu-driven tracking system in that SDK's Unity wrapper (allowing the development of tracking applications without having to know SDK programming).

https://www.youtube.com/watch?v=HspFO87NHeY

Or are you seeking a more modern hand-tracking solution that supports SR300?

closed 2021-05-02 by MartyG-RealSense