realsenseai/librealsense #8823: Turning off projectors to reduce power consumption

rupak-d, 2021-04-13:
Camera Model: D400
Firmware Version: 05.12.10.00
Operating System & Version: Linux
Kernel Version (Linux Only): 5.4.3
SDK Version: 2.43.0
Language: C++

### Issue Description
We have a 3 camera setup and are trying to save power by doing the following -

- Setting the projectors to 25 after 3 seconds of no movement of our platform.
- Turning the projectors off after 10 seconds of no movement of our platform. 
- Projectors turn on to 300 as soon as there is movement.

Is this a good power saving method? Are there values on how much power consumption is reduced by turning off the projectors?

MartyG-RealSense, 2021-04-14:
Hi @rupak-d  If you believe that power-saving is necessary for your particular application then your 3-step power saving mechanism outlined above is reasonable and logical.  

Section 4.1 of Intel's white-paper document about projectors describes how the **Laser Power** value of the 400 Series cameras relates to the power of the projector in **mW**, with the default Laser Power value of '150' representing 150 mW and the maximum of '360' representing 360 mW.

https://dev.intelrealsense.com/docs/projectors?_ga=2.125541386.1877976161.1602709969-1041004708.1589641371#section-4-increasing-range

rupak-d, 2021-04-14:
Are there any issues with switching the laser power that frequently?

MartyG-RealSense, 2021-04-15:
As the Laser Power value would not be changing constantly (for example, multiple times per second), I do not expect it to have a significant effect on the camera.  The camera's laser can potentially degrade over time though.

https://github.com/IntelRealSense/librealsense/issues/3452#issuecomment-471492320

rupak-d, 2021-04-15:
Thanks @MartyG-RealSense!

MartyG-RealSense, 2021-04-21:
Hi @rupak-d  Do you require further assistance with this case, please?  Thanks!

rupak-d, 2021-04-21:
Nope, I'll close it. Thanks for your help!

closed 2021-04-21 by rupak-d

MartyG-RealSense, 2021-04-21:
Thanks for the update @rupak-d - you are very welcome!