realsenseai/librealsense #8824: Projector status in code

rupak-d, 2021-04-13:
Camera Model: D400
Firmware Version: 05.12.10.00
Operating System & Version: Linux
Kernel Version (Linux Only): 5.4.3
SDK Version: 2.43.0
Language: C++

### Issue Description
We have had some cameras on which the projectors had stopped working resulting in bad depth data. It might or might not be a hardware issue on our side. Is there a way to know in code if the projectors are working or not?

MartyG-RealSense, 2021-04-13:
Hi @rupak-d  You could monitor the status of **RS2_OPTION_LASER_POWER** - when it has a value of '0' then the projector is off.

Example code for the control of laser power is provided in the link below, under the 'librealsense2' heading.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

rupak-d, 2021-04-13:
Does that show what the laser power is set to? Or does it show it's current value?

MartyG-RealSense, 2021-04-13:
You should be able to query the value in real-time.  The link below provides example C++ code that prints the current value of **RS2_OPTION_LASER_POWER**

https://github.com/IntelRealSense/librealsense/issues/4923#issuecomment-544816825

rupak-d, 2021-04-14:
How will this differentiate between the laser being at 0 power due to lighting conditions and it not working?

MartyG-RealSense, 2021-04-15:
If the laser is using the **Laser** setting (Setting '1') then it should ignore the lighting conditions and remain on.  The laser would be turned off by the firmware's laser safety mechanism though if the camera's detected projector temperature exceeded 60 degrees C for a period of time.

When the temperature is 60 degrees C or more for several seconds then the Laser Power is first halved by the laser safety mechhanism.  If the temperature continues to exceed 60 degrees after that then the laser is shut off.  The laser will also shut off if the projector temperature falls below 0 degrees C.

I believe that the **Laser Auto** setting (Setting 2) was originally designed to change the emitter status based on lighting conditions, but that setting was deprecated a long time ago.  Currently, in the RealSense Viewer at least, setting Laser Auto mode does not change the Laser Power value and just renders the IR dot pattern projection inactive.

It is a fair point though that checking the Laser Power value may indicate when the value is '0' but would not on its own indicate the reason why it had become '0'.

There was a past case where the IR emitter of a camera showed as enabled but the dot pattern projection was not visible on the infrared image.  In that case, a RealSense team member suggested a method for performing a command-line check in the Viewer for whether there was a hardware error with the projector on a camera.

https://github.com/IntelRealSense/librealsense/issues/7838#issuecomment-759658252

rupak-d, 2021-04-15:
How does rendering the IR dot projection pattern inactive work?

MartyG-RealSense, 2021-04-16:
In camera models such as the D415, the dot pattern can be removed from the image with **color correction**.  The D415 supports a **Visual Preset** camera configuration setting called **Left Imager Color w/o IR Pattern**.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

Point 3 of the section of Intel's camera tunig guide that I have linked to below has further information about color correction.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-the-left-color-camera

This feature was also being worked on for the D455 camera model but I have not seen updated information about that since late 2020.

https://github.com/IntelRealSense/librealsense/issues/7149

closed 2021-04-28 by MartyG-RealSense