realsenseai/librealsense #8825: L515 Depth intrinsic model

anhTuan0712, 2021-04-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | l515 | 
| Firmware Version                   | 01.04 | 
| Operating System & Version |   Win (8.1/10)   | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.36 }                          | 
| Language                            |  C#                       | 
| Segment			|                           | 

### Issue Description
I currently have 2 L515 cameras, 1 of which has problem with intrinsic params, parameters in intrinsic ( These are exactly these parameters: Principal Point, Focal Length) have changed, it's different from when I saw last week.

I'm sure it has been changed, but I can't prove it right now and I don't know how to reproduce the problem.

This camera is used by me and my colleagues, so I'm not sure anyone makes intrinsic changes somehow.
* I used this tool "rs-sensor-control.exe" to see intrinsic parameter.
### Question 
Hi @RealSenseSupport, @dorodnic , @MartyG-RealSense.
1) Is there any way to change intrinsic on L515?
  2) Can intrinsic be altered by any of the following:
  2.1. Due to version Firmware problem?
  2.2. Due to Version SDK problem?
  2.3. Other
 Note: It only happened on one of my cameras, the other was fine.
Please help me to solve this problem.

RealSenseSupport, 2021-04-15:
Hi @anhTuan0712 

Unfortunately without seeing the exact information of the camera in the previous setup and after the new changes, we cannot fully outline what changed the values.  Yes our FW has the ability to make changes to the camera specifics like this.

What FW do you have on the cameras previously and what FW do you have on the camera now?

anhTuan0712, 2021-04-16:
Hi @RealSenseCustomerSupport .
Thank you for the feedback.
step 1: When I buy the camera on FW is: 01.04.01.02, the intrinsic parameter is INTRINSIC1
step 2: My colleagues used the camera (L515)
step 3:  + when I get the camera again, FW: 01.05.00, intrinsic parameter is INTRINSIC2 (INTRINSIC1 #! INTRINSIC2)
             + I tried to re-install FW which is 01.04.01.02, 01.04.01.00 but the intrinsic is still INTRINSIC2  (INTRINSIC1 #! INTRINSIC2)
I'm not sure someone did something in step2

anhTuan0712, 2021-04-20:
Hi @RealSenseSupport, @dorodnic , @MartyG-RealSense.
Please reply to me with this information, I really need it now!

nh-97, 2021-04-20:
I also have the same problem,
 @RealSenseSupport , @dorodnic , @MartyG-RealSense , 
 Please answer the above problem, Thank!

RealSenseSupport, 2021-04-20:
Hi @anhTuan0712 and @nh-97 

Do you have a log of the old intrinsic and the new intrinsic information?  I'm hoping I can get the differences to allow for us to do further investigation.  Can you outline if your colleagues did anything specific to the camera?  Just used the camera normally with the SDK?  They must have updated the FW to 1.5.0.0.  There is no command you can do to force changes at the FW level, so unfortunately I cannot provide anything on that front.  But I do want to investigate further.  Can you also provide the serial number of your cameras that show this behavior?

anhTuan0712, 2021-04-22:
Hi @RealSenseSupport.
## 1.1) Current intrinsic parameters on my camera:
### Resolution 640x480
[image]
### Resolution 1024x768
[image]
## 1.2) Sorry, I don't have info about the old intrinsic.

# 2) Yes, Just used the camera normally with the SDK.

# 3) My colleague used "Intel RealSense Viewer" to update the FW.

# 4) serial number of cameras
[image]

RealSenseSupport, 2021-04-23:
@anhTuan0712 

Thanks for the information.  We will start investigating.

anhTuan0712, 2021-04-26:
Hi @RealSenseSupport.
Please provide me with information as the results of the investigation become available.

anhTuan0712, 2021-05-10:
Hi @RealSenseSupport  !
You have any information for this problem.

RealSenseSupport, 2021-05-12:
Hi @anhTuan0712 

We took a look at the serial number you provided F0264364 along with the intrinsic parameter information you were able to provide and found that the values are the same as what was shipped with this unit.  We do not see a change in the values.

anhTuan0712, 2021-05-14:
Hi @RealSenseSupport.
Sorry for asking for a bit old information, but I want you to confirm this. Which of the following confirmations of yours are correct:
1) Yes our FW has the ability to make changes to the camera specifics like this.
2) There is no command you can do to force changes at the FW level, so unfortunately I cannot provide anything on that front.
[image]

anhTuan0712, 2021-05-14:
> Hi @anhTuan0712
> 
> We took a look at the serial number you provided F0264364 along with the intrinsic parameter information you were able to provide and found that the values are the same as what was shipped with this unit. We do not see a change in the values.

Thanks for your help, but like I said the "intrinsics" have changed since I first got the Camera. I hope you will continue to investigate this matter.

RealSenseSupport, 2021-05-14:
Hi @anhTuan0712 

If we run across the problem again then we will be able to investigate further.

Is there anything else on this matter?

anhTuan0712, 2021-05-17:
> Hi @RealSenseSupport.
> Sorry for asking for a bit old information, but I want you to confirm this. Which of the following confirmations of yours are correct:
>
> 1. Yes our FW has the ability to make [...]

Hi @RealSenseSupport. 
Please confirm this question.

RealSenseSupport, 2021-06-09:
Hi @anhTuan0712 

To confirm, internally we have the ability to modify camera specifications as we are able to build FWs to do this.  We also have internal engineering tools that have register level capabilities so commands with our public level tools are not available.

I hope this answers your final questions.

I want to let you know that we have not made such modifications to intrinsic information.

closed 2021-06-18 by RealSenseSupport