realsenseai/librealsense #8826: D435i on Raspberry Pi 3b+, RuntimeError: Couldn't resolve requests

stino78, 2021-04-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | 5.12.2.100 | 
| Operating System & Version |  Raspberry Pi OS 2020-08-20-raspios-buster-armhf-lite  | 
| Kernel Version (Linux Only)    |             5.4.51                            | 
| Platform                                 | Raspberry Pi 3b+  |
| SDK Version                            |  { 2.39.0 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {HMD }                          | 

### Issue Description
I have a system perfectly working, then I copied it, hardware and software, and it does not work.

In my code, I create 2 pipelines, one for the images and one for the IMU, for asynchronous reading. The device can stream perfectly the following data:

- depth at 480 x 270, 6fps

- color at 640 x 480 at 6 fps

- gyro at 200Hz

- acceleration at 250Hz

 

 

I then acquired 4 new Realsense D435i cameras, and they give an error at runtime:

pipeline_IMU.start(config_IMU) RuntimeError: Couldn't resolve requests

 

Launching rs-enumerate-devices.exe it shows the depth sensor only, which is probably the problem.

I tried to update the firmware to a new version, but the problem was not solved. The serial number of one of the new devices is: 017322070770

Note that
- the Raspberry Pi OS is headless, so I cannot see use the Realsense Viewer

- if I connect the D435i on a windows machine it shows correctly on the Realsense Viewer
 

Any suggestions?

 

Thank you,

 

Agostino

MartyG-RealSense, 2021-04-14:
Hi @stino78  The resolution and FPS choices for depth and color should be compatible with the modes available on a USB 2 connection.  Your accel and gyro settings look fine too.  I note that you were able to use the settings successfully when connecting the D435i to a Windows machine.

[image]

If the cameras are identified as D435i when connected to the RealSense Viewer on the Windows machine then there is likely not a physical hardware problem with the D435i, as the firmware driver should identify the model as D430i if the RGB sensor's presence could not be detected.

Could you try the cameras with the official 1 meter USB-C cable that was supplied with the camera if you have not done so already, please?  You should not need a dedicated USB 2 cable, as when the official cable is plugged into a USB 2 port then the camera should simply operate in USB 2 mode.

stino78, 2021-04-14:
Hi MartyG, thank you for the help!

I can try changing the cable for sure.

BUT

I assembled 4 copy devices and I tested them with the "first" D435i, always the same. 
Then the Realsense D435i was changed, and it is the only change, and the error appeared.
Unluckily, the devices are in another country, so I cannot put the first D435i back in and see if it works.

I just realized that I mounted a slightly older firmware, the 5.12.15.0 instead of the one that I tested, the 5.12.2.100. In the release notes I don't see any problems, but you might know something, and I will test this before changing the cable. 

Will let you know.

MartyG-RealSense, 2021-04-14:
When you mention firmware 5.12.15.0, do you mean **5.11.15.0** please, as the most recent firmware version at the time of writing this is **5.12.12.100**

If you were using 5.11.15.0 with the **2.39.0** RealSense SDK version then that could be an issue, as the recommended minimum firmware version for that SDK is **5.12.8.200**.

I recall a case where cameras were tested in an office in the US and worked fine and were shipped to an exhibition center in another country, but then had problems after they were set up.  It was thought that the different mains electricity system in the country that the cameras had been sent to might be a factor.

Another factor with the cameras that were experiencing the problems was that long USB extension cables were being used (30ft of cable with 'active repeater' components in them to boost the signal).  Are you using long cables please?

https://github.com/IntelRealSense/librealsense/issues/6805

stino78, 2021-04-14:
Sorry for messing up with numbers, here's the correct ones. Note that the two software configuration have been tested on two different D435i, not sure if it matters...

WORKING CONFIGURATION
Serial # 937622071937
SDK 2.39.0
Firmware 5.11.15.0

NOT WORKING CONFIGURATION
Serial # 017322070770
SDK 3.39.0
Firmware 5.12.2.100

I will try mounting the recommended firmware.

stino78, 2021-04-14:
Sorry, using a 1.5 meters cable.

MartyG-RealSense, 2021-04-14:
Okay, thanks very much for the information.  Please do update here when you have updated the firmware.  Thanks!

stino78, 2021-04-16:
Ok, problem solved.

The recommended firmware for SDK 2.39.0 is 5.12.12.100, but it is not working on my system.

After some tests, the firmware that is working is the **5.12.2.100**

MartyG-RealSense, 2021-04-16:
Thanks very much @stino78 for the update - it is great to hear that you found a solution that works for you!

closed 2021-04-22 by MartyG-RealSense