realsenseai/librealsense #8827: D435i IR illuminator dont emit light

DuyguSerbes, 2021-04-14:
Hello,

I am using D435i camera and SDK2.0. But when I started IR stream, the illuminators dont start to work and emit the IR lights. How can I initiate them?

MartyG-RealSense, 2021-04-14:
Hi @DuyguSerbes  As a starting point in investigating your case, may I ask please if you are using the **RealSense Viewer** tool?  And if you are, is the **Emitter Enabled** option set to the **Laser Auto** setting?

[image]

If you are using the Viewer and this option is set to Laser Auto then please change it to **Laser**, otherwise the IR emitter will be inactive.

DuyguSerbes, 2021-04-14:
Yes, I am using Realsense Viewer. But previously, I used to SR305 and I have one infrared option as Infrared 1 currently with D435i. Should I update the Viewer bu uninstalling and reinstalling?

MartyG-RealSense, 2021-04-14:
The D435i and SR305 have different camera technologies ('stereo depth' for D435i and 'coded light' for SR305), so they cannot be compared directly.  

If your D435i has the Emitter Enabled option set to 'Laser' then please enable the Stereo Module with 'Infrared 1' selected and point the camera at your hand to see whether dots appear on the hand on the on-screen infrared image.

closed 2021-05-02 by MartyG-RealSense