realsenseai/librealsense #8828: get 3D ply file without ros

hyuDev, 2021-04-14:
hi, I want to get ply file or Pointcloud data.

I get a rgb image and depth image from example code,unfortunately I can get a ply file but it look like snapshot.
I want to scan a fully object.

some people comment me, using ROS bag.
My env can't install ROS so, I want to 3d pointcloud file without ROS.
Is it possible?

MartyG-RealSense, 2021-04-14:
Hi @hyuDev  Yes, it is possible to obtain a 360 degree ply or pointcloud scan of an object or location without ROS.  Could you provide the following information please.

1.  Are you using a single RealSense camera or multiple cameras.
2. If you are using a programming language (C++, Python, C# etc), which language are you using?
3. Is installing commercial software on your system an option?

sonumathur, 2021-04-14:
Hi @MartyG-RealSense
How we can obtain 360 degree ply using python.

closed 2021-04-26 by MartyG-RealSense