realsenseai/librealsense #8829: D435 realsense-viewer normal operation , but the sample program failed to run

liu-jiaming, 2021-04-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (2.44.0) | 
| Operating System & Version |   {Ubuntu 16.04.7 (64bit)  | 
| Kernel Version (Linux Only)    |  ( 4.4.38)                                         | 
| Platform                                 |  Manifold - 2G (SZ DJI Technology Co.,Ltd. ) (CPU: NVIDIA Jetson TX2 ) |
| SDK Version                            |  { legacy / 2.44.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I installed the SDK according to the instructions of [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)
 and executed the command realsense-viewer. It seems that the program runs normally and the depth image can be displayed normally. When I execute examples/rs-align, the following prompt appears, other sample programs also have this error

RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x662a70):     Frame didn't arrive within 15000

MartyG-RealSense, 2021-04-14:
Hi @liu-jiaming   Do you have the RealSense Viewer streaming at the time when you launch the example programs, please?  If you do, then please try stopping the streaming in the Viewer and then launching the example.

The opposite is also true.  If you are currently using an example program then close it before enabling streaming in the Viewer.

The example programs are not dependent on the Viewer and can run without the Viewer being launched.

liu-jiaming, 2021-04-14:
> Hi @liu-jiaming Do you have the RealSense Viewer streaming at the time when you launch the example programs, please? If you do, then please try stopping the streaming in the Viewer and then [...]

I turned it off

MartyG-RealSense, 2021-04-14:
Thank you.  Next, could you try the command **rs-depth-quality** in the Ubuntu terminal to see whether the Depth Quality Tool can be launched and used successfully like the Viewer can, please?

liu-jiaming, 2021-04-21:
> Thank you. Next, could you try the command **rs-depth-quality** in the Ubuntu terminal to see whether the Depth Quality Tool can be launched and used successfully like the Viewer can, please?

There is something wrong with the system. Recompiling and installing. When I execute the "./scripts/patch-realsense-ubuntu-L4T.sh" script, the following prompt appears, and then I skip this step. I'm not sure what this prompt means

Jetson Board (proc/device-tree/model): quill
Jetson L4T version: 28.2.1
Unsipported JetPack revision 28.2.1 aborting script

liu-jiaming, 2021-04-21:
> Thank you. Next, could you try the command **rs-depth-quality** in the Ubuntu terminal to see whether the Depth Quality Tool can be launched and used successfully like the Viewer can, please?

the command rs-depth-quality in the Ubuntu terminal to see the Depth Quality Tool can be launched and used successfully like the Viewe.
but when I run the rs-capture command, the following prompt appears

21/04 02:39:41,100 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,229 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,358 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,488 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,615 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,743 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:41,877 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
 21/04 02:39:42,000 ERROR [548205945296] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: \ufffd
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x9e61c0):
    Frame didn't arrive within 15000

MartyG-RealSense, 2021-04-21:
I note the error in your log **Jetson L4T version: 28.2.1  Unsipported JetPack revision 28.2.1 aborting script**.  I also note that you built from source with the Native Backend, since you used **scripts/patch-realsense-ubuntu-L4T.sh**

Your L4T version 28.2.1, which equates to JetPack 3.3, is quite old now.  The documentation for this installation method states that the installation script was verified with JetPack 4.2.3 [L4T 32.4.3], 4.3 [L4T 32.3.1] and 4.4 [L4T 32.2.1].

It is also worth mentioning that SDK 2.43.0 added support for Jetpack 4.4.1+4.5 (32.4.4 + 32.5)

Would it be possible to upgrade your L4T version to one of the above see whether it makes a positive difference?

closed 2021-05-03 by MartyG-RealSense