realsenseai/librealsense #8830: Changing depth values in D455  due to USB cable.

rghl3, 2021-04-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.12.12.100 | 
| Operating System & Version | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  5.4.0-70                                        | 
| Platform                                 | PC|
| SDK Version                            | 2.43.0 | 
| Language                            |  C++|  

### Issue Description

Hi, I have a d455 camera that's fit at a height of 100cm from the floor, facing downwards.

I am using a USB 3.1 cable with screws [similar to this](https://www.amazon.com/Newnex-Cable-Standard-Type-C-Locking/dp/B01LWX26WR/ref=cm_cr_arp_d_product_top?ie=UTF8) to connect my pc to the camera.

When I screw the cable, the depth value decreases by around 2cm ie)98cm. But when I unscrew again, the depth value comes again as 100cm. Why does the depth value change because of the screws?

MartyG-RealSense, 2021-04-14:
Hi @rghl3   I am not aware of any past case where using a screw-lock USB cable has affected depth measurement in this way.

Is there high vibration at the mounting point of the camera, and if so then might that vibration be being transferred to the camera if the cable is locked to it with a skin-tight fit by the screw-locks? 

Examples of vibration being transferred through a mounting point to the camera include cameras on agricultural vehicles travelling on rough ground or flying drones that transmit vibration through the frame of the drone.

rghl3, 2021-04-15:
HI @MartyG-RealSense.
The camera is kept static. So there are no vibrations at the mounting point.

MartyG-RealSense, 2021-04-15:
I do not have a screw-lock USB cable to test with to verify your findings.  I will pass your experience on to Intel though.  Thanks very much for the report!

MartyG-RealSense, 2021-04-15:
Hi @rghl3  Could you provide a picture of how the camera is mounted, please?

rghl3, 2021-04-16:
Hi @MartyG-RealSense,
I have unmounted the camera. I will send a picture once I re-assemble. But I could reproduce the same issue by just fixing the screws on either side and using default USB cable that comes with d455.

MartyG-RealSense, 2021-04-16:
Thanks very much for the further information @rghl3 - I look forward to your image.

closed 2021-05-03 by MartyG-RealSense