realsenseai/librealsense #8831: How to reconnect L515

mingyangpeng, 2021-04-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515 } | 
| Firmware Version                   | (01.05.05.00) | 
| Operating System & Version |   { Linux 18.04 }  | 
| Kernel Version (Linux Only)    |  (4.9.201-tegra)  | 
| Platform                                 |   Jetson nano   | 
| Language                            |  {C++ }                          |  

### Issue Description 

Hello, when l515 is not working properly, I try to reset it, 
```
       rs2::context ctx;
       auto devices = ctx.query_devices()[0];
       devices.hardware_reset(); 
```

but it report error:
```
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state 

```

and I try to us rs-distance:
```
There are 1 connected RealSense devices.
rs_error was raised when calling rs2_create_device(info_list:0x55c2641330, index:0):
    failed to set power state
```
 
I can only restart the system or pull down the USB and reconnect, how can I restart l515 by code?

mingyangpeng, 2021-04-18:
hello, who can help me?

mingyangpeng, 2021-05-18:
@RealSenseSupport 
Is there any way to solve it now?

RealSenseSupport, 2021-05-18:
Hi @mingyangpeng 

So is the program closes unexpectedly not related to the L515 camera?  Is it related to the jetson nano itself being unstable?

Normally the hardware reset should work without issues.  But if there is something wrong with the USB enumeration or the connection to the camera then the hardware reset can fail to actually reset.

Have you tried just a basic hardware reset to the L515 before a unexpected program closure?

mingyangpeng, 2021-05-21:
@RealSenseSupport 
Thanks for the help.
What I want to ask is that   the current program exits unexpectedly, call l515 again. How to release l515 occupied by the previous program

RealSenseSupport, 2021-06-10:
@mingyangpeng 
Were you able to get past any issues you've been seeing?

07hokage, 2021-06-17:
i had the similar issue while operating multiple D435 cameras. when the pipelines are closed properly in case of exceptions, i was able to do hardware reset using similar code as above. But if **either pipeline is not closed properly or if more than 2 devices are running already**, then i got the error "failed to set power state" when i call `context.query_devices()`

mingyangpeng, 2021-06-17:
I don't have a good way at present. I'm waiting for the official solution. @07hokage @RealSenseSupport

07hokage, 2021-06-17:
@mingyangpeng  did you close the realsense pipeline when your scripts ends unexpectedly? if it was closed properly (like `pipeline.stop()` )then i guess you should be able to do hardware reset. usually the `context.query_devices()` hangs when there is already an open pipeline as far as i have experienced.

mingyangpeng, 2021-06-18:
@07hokage As you said, It didn't close the realsense pipeline when my scripts ends unexpectedly.

RealSenseSupport, 2021-07-01:
@mickeyouyou @07hokage , could you please give the latest SDK a try and see if the problem still persists with 2.48 release?
https://github.com/IntelRealSense/librealsense/issues/9315
Thank you.

mingyangpeng, 2021-07-06:
@RealSenseSupport 
I'm sorry to  respond you so late, according to your advice, It works well with 2.48 release, thank you!

closed 2021-07-06 by mingyangpeng