realsenseai/librealsense #8834: RealSense error calling rs2_pipeline_start(pipe:0x57bfa0)

sandra-mf, 2021-04-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |  Linux   | 
| Kernel Version (Linux Only)    |  4.14.34-rt27-aimx6                             | 
| Platform                                 | emPC-A/iMX6   |
| SDK Version                            |  2.44.0                        | 
| Language                            |  C                         | 
| Segment			|  Robot                          | 

### RealSense error calling rs2_pipeline_start(pipe:0x57bfa0) 

Hello,

I just compiled the librealsense libraries on my emPC-A/iMX6, but the examples seem not find my D435. I am using the original USB cable, although the USBs on the emPC-A/iMX6 are 2.0.

When I connect the camera, I get this with dmesg | grep 8086:
[32134.622781] usb 2-1.1: New USB device found, idVendor=8086, idProduct=0b07

But when I run pipe.start(); I get this error:
RealSense error calling rs2_pipeline_start(pipe:0x57bfa0):
    No device connected

Can you help me?

MartyG-RealSense, 2021-04-15:
Hi @sandra-mf   The **emPC-A/iMX6** is an industrial PC with a dedicated Linux kernel.  I would therefore recommend building the librealsense SDK from source code with CMake using the build flag **-DFORCE_RSUSB_BACKEND=true** to bypass the kernel.

The RSUSB installation method installs librealsense over an internet connection and is not dependent on Linux versions or kernel versions and does not require kernel patching.  The link below has instructions for performing this type of installation.

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

sandra-mf, 2021-04-15:
Hi !

Thank you very much for your answer, I recompiled it with the -DFORCE_RSUSB_BACKEND=true flag and now I get a different error :

15/04 16:41:10,109 ERROR [1759921232] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state

any idea?

MartyG-RealSense, 2021-04-15:
Are you using a single camera or more than one camera on the computer, please?

sandra-mf, 2021-04-15:
Just one camera, and I think the problem was that I was starting the pipeline twice. Sorry.
Now it is working
Thank you very much for the support!

MartyG-RealSense, 2021-04-15:
No problem at all, I'm very pleased to hear that you found the solution.  Thanks for the update!

closed 2021-04-22 by MartyG-RealSense