realsenseai/librealsense #8835: D400

L-xn, 2021-04-15:
用D400系列做手指追踪有进展了吗？我正在做我的本科毕业设计！我需要识别手部22个关节点。但是nuitrack不能完成。或许只能依靠OpenCV。有关于OpenCV在这一方面的资料或者代码吗？我真诚地恳求一些指导！谢谢！

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Platform                                 |vs / etc..  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++/python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-04-15:
Translation: "Is there any progress in finger tracking with D400 series?  I am doing my undergraduate graduation project!  I need to identify 22 joint points of the hand.  But Nuitrack cannot be completed.  Maybe you can only rely on OpenCV.  Is there any information or code about OpenCV in this regard?  I sincerely beg for some guidance!  Thank you!"

Hi @L-xn  If you would prefer a finger trackig solution that has already been created then the link below leads to a recent Python project that supports 22 finger joints.

https://github.com/AlextheEngineer/Ego3DHands

There is also a tutorial for creating a 22 point finger tracking system in OpenCV in the link below.

https://learnopencv.com/hand-keypoint-detection-using-deep-learning-and-opencv/

****************

如果您希望使用已經創建的手指跟踪解決方案，那麼下面的鏈接將導致一個最近的Python項目，該項目支持22個手指關節。

https://github.com/AlextheEngineer/Ego3DHands

在下面的鏈接中，還有一個用於在OpenCV中創建22點手指跟踪系統的教程。

https://learnopencv.com/hand-keypoint-detection-using-deep-learning-and-opencv/

L-xn, 2021-04-15:
> Translation: "Is there any progress in finger tracking with D400 series? I am doing my undergraduate graduation project! I need to identify 22 joint points of the hand. But Nuitrack cannot be completed. Maybe you can only rely on OpenCV. Is there any information or code about OpenCV in this regard? I sincerely beg for some guidance! Thank you!"
> 
> Hi @L-xn If you would prefer a finger trackig solution that has already been created then the link below leads to a recent Python project that supports 22 finger joints.
> 
> https://github.com/AlextheEngineer/Ego3DHands
> 
> There is also a tutorial for creating a 22 point finger tracking system in OpenCV in the link below.
> 
> https://learnopencv.com/hand-keypoint-detection-using-deep-learning-and-opencv/
> 
> 你好 @L-xn
> 
> 如果您希望使用已經創建的手指跟踪解決方案，那麼下面的鏈接將導致一個最近的Python項目，該項目支持22個手指關節。
> 
> https://github.com/AlextheEngineer/Ego3DHands
> 
> 在下面的鏈接中，還有一個用於在OpenCV中創建22點手指跟踪系統的教程。
> 
> https://learnopencv.com/hand-keypoint-detection-using-deep-learning-and-opencv/

Very grateful for the quick reply! If opencv recognizes the joint points, whether D415 can be used to measure the frequency and angle of finger movement. Thank you!

MartyG-RealSense, 2021-04-15:
If you are going to be using OpenCV then **OpenPose** might meet your needs for finger angle tracking as that is based upon OpenCV too.

https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose

If you need to measure movement, the link below about tracking a small object might provide useful ideas.

https://github.com/IntelRealSense/librealsense/issues/4175

*************************

如果您要使用OpenCV，則OpenPose可能也滿足您對手指角度跟踪的需求，因為它也是基於OpenCV的。

https://github.com/gregorsamsa183/RealTime3DPoseTracker-OpenPose

如果需要測量運動，下面有關跟踪小物體的鏈接可能會提供有用的想法。

https://github.com/IntelRealSense/librealsense/issues/4175

MartyG-RealSense, 2021-04-21:
Do you require further assistance with this case please?  Thanks!

在這種情況下，您需要其他幫助嗎？ 謝謝！

closed 2021-04-27 by MartyG-RealSense