realsenseai/librealsense #8836: Device_hub doesn't detect connect events

daziFR, 2021-04-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                   D435                             |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 |  PC |
| SDK Version                            | 2.44.0              | 
| Language                            |  C/C++                          | 
| Segment			| others                         | 

### Issue Description
Hi. 

We encounter the device_hub issue, which doesn't detect plug-in events.

How to reproduce:

1. Start app with code:
    ```
    rs2::context context; 
    rs2::device_hub hub(context);
    hub.wait_for_device();
    ```
2. Plug-in camera
3. Wait_for_device doesn't return, it blocks all the time.

From the code, it seems that device_watcher is never started, just registered. Any comments on this? 

Thank you.

BR,
Danijel

MartyG-RealSense, 2021-04-15:
hi @daziFR   If you are aiming to detect a camera insertion event, this could perhaps instead be handled with the **set_devices_changed_callback** function.

https://support.intelrealsense.com/hc/en-us/community/posts/360046766314/comments/360011536654

daziFR, 2021-04-15:
Yes if I use **set_devices_changed_callback** it works as expected. But isn't originally intended for the device_hub to handle such cases?

`    device_hub::device_hub(std::shared_ptr<librealsense::context> ctx, int mask, int vid,
                           bool register_device_notifications)
        : _ctx(ctx), _vid(vid),
          _device_changes_callback_id(0),
          _register_device_notifications(register_device_notifications)
    {
        _device_list = filter_by_vid(_ctx->query_devices(mask), _vid);

        auto cb = new hub_devices_changed_callback([&,mask](rs2::event_information&)
                   {
                        std::unique_lock<std::mutex> lock(_mutex);

                        _device_list = filter_by_vid(_ctx->query_devices(mask), _vid);

                        // Current device will point to the first available device
                        _camera_index = 0;
                        if (_device_list.size() > 0)
                        {
                           _cv.notify_all();
                        }
                    });

        //_device_changes_callback_id = _ctx->register_internal_device_callback({ cb, [](rs2_devices_changed_callback* p) { p->release(); } });
        // CHANGE
        _ctx->set_devices_changed_callback({ cb, [](rs2_devices_changed_callback* p) { p->release(); } });
`

MartyG-RealSense, 2021-04-15:
The **device_hub** documentation's information about the **wait_for_device** sub-option would seem to suggest that it keeps blocking because it does not detect a connected device.  I wonder whether it means 'connected' in terms of an insertion event, not a camera that is already plugged in when the program is launched.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__hub.html#a36b20936725a27bf49887fbccf405fd7

The SDK's **rs_sensor_control** example program provides an illustration of using device_hub.

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L36

MartyG-RealSense, 2021-04-21:
Hi @daziFR  Do you require further assistance with this case, please?  Thanks!

daziFR, 2021-04-21:
Thanks @MartyG-RealSense. For now this topic can be closed. Maybe just to notice realsense guys to check/modify documentation for device_hub, as it doesn't work the way it suggests (handling connect/disconnect events, it handles already connected device).

closed 2021-04-21 by daziFR

MartyG-RealSense, 2021-04-21:
Thanks very much @daziFR for the update!  

If you have feedback regarding documentation, could you create a new case with those details please?  I can then give that case the **Documentation** label to mark it for future consideration by the RealSense team members responsible for documentation and leave the case open in the meantime.