realsenseai/librealsense #8839: [IDEA] Make the kernel patches more robust/future-proof

thomas-slamcore, 2021-04-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.8                                     | 
| Platform                                 | PC |
| SDK Version                            |  v2.44                         | 
| Language                            |  -                    | 

### Issue Description

Over the last months I´ve used the kernel 5.4 patches on different kernels and different distributions. I´ve noticed that while there have been no changes to the code that is being patched, the patchset keeps breaking. This seems to be exclusively down to new video formats being added to the upstream kernels or new UVC device quirks. Since the obvious thing when adding new items to a list is to append them (in some cases it´s mandatory in order not to break compatibility) all the changes happen at the end of the lists. Thus whenever someone contributes a new video format or adds a new device quirk, the patches stop applying cleanly.

My idea is to modify the patches to prepend to the lists rather than appending. This would obviously not be a good look if the intent were to upstream the patches but it does seem to make the patches more robust. Indeed, I´ve been able to use the exact same patches on Ubuntu 18.04/kernel 5.4, Ubuntu 20.04/kernel 5.4, Ubuntu 20.04/kernel 5.8 and Archlinux/kernel 5.11.

The patching scripts still need to be made aware of the new HWE 5.8 branch in order to work. The second part of my proposal is to assume that an unknown Ubuntu 20.04 kernel is from a new HWE branch, and attempt to apply the existing patches. If the patches apply without error (and they do), then proceed with just a warning.

I´m aware that support for kernel 5.8 on Ubuntu 20.04 has been added recently to the development branch but this idea could potentially serve a stopgap whenever Canonical release a new HWE kernel during that period of time before Realsense release official support for it.

The same concept could be used on the librealsense-dkms package although I don´t know where it´s sources live.

MartyG-RealSense, 2021-04-16:
Thanks very much @thomas-slamcore for your detailed ideas!  I have highlighted this case to Intel and also marked it with an **Enhancement** tag as a reminder that this case should be kept open whilst your Pull Request https://github.com/IntelRealSense/librealsense/pull/8840 is active.

RealSenseSupport, 2021-08-25:
This issue has been tagged and being tracked internally and therefore this ticket will be closed.

closed 2021-08-25 by RealSenseSupport

drzraf, 2021-09-03:
@RealSenseSupport : Don't you think the *real* fix would be to upstream these patches and been done with them once for all?
Where are the inclusion request being tracked?