realsenseai/librealsense #8841: Device is not recognized as connected

EZIO7000, 2021-04-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 2.40.0 | 
| Operating System & Version |   Ubuntu 20.4, Windows 10  |                                   

### Issue Description
When I went through installation guide on main page for Ubuntu and open realsense-viewer I saw that I can not turn on Depth Sensor and Motion Sensor because of the error :

"Backend in rs2_open_multiple(sensor:0x7fa8ac01bca0, profiles:0x7fa8ac41c490, count:2):
xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error" 

so I decided to upgrade firmware, and then I got message that program was unable to connect to device, and after that I can not connect to my camera anywhere. When I plug it in in win10 it doesn't appear in device manager as new device (it doesn't appear at all), and the same on Ubuntu, no new behaviours  in dmesg when I plug in to my usb port. 
The port is working and I tried it on multiple computers with the same result.

EZIO7000, 2021-04-19:
@RealSenseSupport

RealSenseSupport, 2021-05-03:
Hi @EZIO7000 

Sorry to hear you're running into this type of behavior.  

Couple of questions:
- Does behavior change based on the orientation of how you plug the USBC connector into the camera?
- Can you provide images of device manager when you connect the camera to the two orientations of the USBC connector?
- Just to confirm are you using the USB cable that came with the camera?

You can open a ticket here to connect 1:1 with one of our agents:  https://support.intelrealsense.com/hc/en-us/requests/new

EZIO7000, 2021-05-06:
1) Orientation doesn't change the behaviour.
2) It's the same in the different orientations. No new device detected.
[image]
[image]
3) I use the USB cable that came with the camera. As I said, it worked before with the same cable.

RealSenseSupport, 2021-05-06:
Hi @EZIO7000 

Thanks for the images and the information.  I think the suggestion right now would be to look into RMA the camera.  Looking into where you purchased the camera from and request RMA.

closed 2021-06-09 by RealSenseSupport