realsenseai/librealsense #8842: Converting an rs2::frameset to an std::vector

rolandgvc, 2021-04-15:
### Issue Description
I would like to convert a streamed infrared image into an std::vector or fixed size std::array without using cv::Mat. Would that be possible?

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |  Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |   5.4.0   | 
| SDK Version                            |    2     | 
| Language                            |  C++  |

MartyG-RealSense, 2021-04-16:
Hi @rolandgvc  Could you provide more information please about what you are trying to achieve with your conversion?

For example, a script in the link below describes how to retrieve RGB pixels that correspond to the infrared image.

https://github.com/IntelRealSense/librealsense/issues/4783

Is that the kind of process you had in mind or a different goal?  Thanks!

rolandgvc, 2021-04-16:
Thanks for your response @MartyG-RealSense. I want to transmit the streamed images over DDS as std::vector / std::array messages. I would like to be able to convert the video_frame data structure into an std container as efficient as possible. Have you dealt with this before?

MartyG-RealSense, 2021-04-16:
I have not seen a scenario like the described one before, unfortunately.

If you are able to use PCL in your project, the link below describes a method for converting a PCL cloud to a vector of Point3f points using a "for" loop.

https://stackoverflow.com/questions/53781038/quick-conversion-of-pclpointcloudpointxyz-to-stdvectorcvpoint3f-using

So if your project was able to generate a PCL pointcloud using the RealSense SDK then that could potentially provide the data to convert to std::vector

rolandgvc, 2021-04-20:
For example, getting the raw data of a infrared stream in a cv::Mat would go something like this:
```c++
rs2::pipeline rs_pipe;
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1, 640, 480, RS2_FORMAT_Y8, 15);
cfg.enable_stream(RS2_STREAM_INFRARED, 2, 640, 480, RS2_FORMAT_Y8, 15);
rs2::pipeline_profile rs_profile = rs_pipe.start(cfg);
      
rs2::frameset frames = rs_pipe.wait_for_frames();
auto rs_left = frames.get_infrared_frame(1);
cv::Mat cv_left = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)rs_left.get_data());
```
What would be the equivalent for an std::vector?

MartyG-RealSense, 2021-04-20:
I'm not certain that the links below that feature **std::vector** are what you're seeking, but I will share them in case they are helpful.

https://github.com/IntelRealSense/librealsense/issues/7010

https://github.com/IntelRealSense/librealsense/issues/937

closed 2021-05-03 by MartyG-RealSense