realsenseai/librealsense #8843: LNK2019 problems when importing example.hpp

Rad0n1, 2021-04-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            | 2.43.0  | 
| Language                            |  {C/C++  | 
| Segment			|  others  | 

Hello realsense team,

This question might be asked on the wrong platform and possibly is very simple to solve, but I am not the most experienced in C/C++ yet so I dont know what to do with it.

I want to import the example header that is placed in the examples. I wanted to do this in my own project but that gives around 32 LNK2019 errors of an unresolved external symbol (I will paste the error code below). Can you guys help me or should I ask this question on a different forum instead.

I am sorry for possibly being lost in the wrong place

Kind regards,
Jorik

P.S.
Error LNK2019	unresolved external symbol __imp__glMatrixMode@4 referenced in function "void __cdecl draw_pointcloud(float,float,struct glfw_state &,class rs2::points &)" (?draw_pointcloud@@YAXMMAAUglfw_state@@AAVpoints@rs2@@@Z)	ProjectName	D:\GitRepos\ProjectName\ProjectName\ProjectName.obj	1

MartyG-RealSense, 2021-04-16:
Hi @Rad0n1  The LNK2019 problem of another RealSense user was solved by following a guide for setting up a new RealSense project with Visual Studio.  That user found that their problem had been caused by not adding the path of **glfw3.lib** in the correct place.  It looks as though you have a similar issue with glfw.

https://github.com/IntelRealSense/librealsense/issues/7106#issuecomment-678922995

Rad0n1, 2021-04-19:
Hey MartyG thank you so much for your response I solved it! I looked at the other post and I found that the lib was linked correctly (installed it through VS2019) however I had no clue about the props files. After adding the prop files to the project it ran fine :)

Sorry for getting back so late, thank you for helping me solve the problem

closed 2021-04-19 by Rad0n1

MartyG-RealSense, 2021-04-19:
No problem at all, I'm pleased that I could help.  Thanks for the update!