realsenseai/librealsense #8845: D450 (bundle) imu availability ?

TheBricktop, 2021-04-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d4550 bundle version } | 
| Firmware Version                   | (newest) | 
| Operating System & Version |   {Win (8.1/10) / Android  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson   |
| SDK Version                            | 2                         | 
| Language                            |  {C/C#/opencv/python/unity }                          | 
| Segment			|  {Smartphone/VR/AR}                          | 

### Issue Description
Does the 450 bunde ( caseless version ) sensor features the same IMU as the cased version of the d455 ?
Is it compatible with android librealsense library?

MartyG-RealSense, 2021-04-16:
Hi @TheBricktop  The **Depth Module** versions of RealSense cameras do not feature the IMU that the IMU-equipped cased cameras (D435i and D455) have, as their purpose is dedicated to depth processing.  The diagram below illustrates which sensors are provided on the D450 module.

[image]

TheBricktop, 2021-04-16:
Hey @MartyG-RealSense I've noticed on some reseller sites that the companion d4 v3 visual processor for d450 actually has imu integrated as is in the spreadsheet from June 2020.

MartyG-RealSense, 2021-04-17:
You are correct, the data sheet document for the 400 Series cameras has a footnote mention at the bottom of page 20 of the current edition.  Thanks very much for highlighting this feature!

[image]

TheBricktop, 2021-04-17:
I think that this somewhat should be mentioned on the Intel store site beneath the D4 board for d450 module .

MartyG-RealSense, 2021-04-17:
The RealSense support forums are not involved in the RealSense online store.  I will certainly remember the information for future enquiries of this type though - thanks very much.

closed 2021-04-23 by TheBricktop