realsenseai/librealsense #8847: Failed to build librealsense - Invalid command line switch for "cmd.exe" (PATH)

lachlansleight, 2021-04-17:
I'm trying to get the nodejs wrapper up and running, and after dealing with a series of problems, I now have one that I'm not sure how to fix.

When running `npm i node-librealsense` I get the following error which fails the build (newlines added for readability):

```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(240,5): 
error MSB6001: Invalid command line switch for "cmd.exe". System.ArgumentException: Item has already been added. Key in dictionary: 'Path'  Key being added: 'PATH' 
[C:\Users\MyUserName\PathToProject\node_modules\node-librealsense\librealsense\build\ZERO_CHECK.vcxproj]
```

Things I've done:

  * Uninstalled and reinstalled Visual Studio Community 2019 with .NET and C++ workflows, and added the MsBuild directory to PATH
  * Uninstalled and reinstalled the newest version of cmake (3.20.1) and ensured that it's in PATH
  * Installed the newest version of the realsense SDK (2.44.0.3056)

I'm running Windows 10, version 20H2 build 19042.928

I don't normally do C++ development so I'm pretty far out of my depth here. I'd never even heard of cmake or MsBuild before today so I don't really know where to start looking to fix this problem. Especially because I get the vibe that the npm package is doing a lot of things under the hood that I don't have much control over.

I've attached the console output when running `npm i node-librealsense`, as well as the log file that was generated.

Thanks in advance for any help!

[error-log.txt](https://github.com/IntelRealSense/librealsense/files/6328818/error-log.txt)
[console-output.txt](https://github.com/IntelRealSense/librealsense/files/6328819/console-output.txt)

MartyG-RealSense, 2021-04-17:
Hi @lachlansleight   Could you try the npm installation method in the link below please?

https://github.com/IntelRealSense/librealsense/issues/7439#issuecomment-708503008

lachlansleight, 2021-04-19:
Hi Marty,

I tried that, but I don't think I really understand exactly what you're asking. I try running that command in my nodejs project, and nothing happens (since there's nothing in my package.json at that point). I tried a few permutations:

Attempt 1:

  * Built the C++ SDK with Release configuration type
  * Ran the command you mentioned in wrappers/nodejs folder, and was told that the script couldn't find the dll in "build/Release" folder
  * Renamed the build folder that was generated by visual studio to the path that the vs configurations seems to want (build/Release), then ran the command again, which seemed to do something
  * Added that package using npm's file dependency system `"dependencies": { "node-librealsense": "file:C:/blahblahblah/wrappers/nodejs" }`
  * Ran `npm i` inside my project folder (not the SDK folder) - which seemed to import some stuff into node_modules, but fails with the following error (I get the same error when running `npm i --vs-configuration=Release`)

```fatal error C1083: Cannot open include file:
'librealsense2/rs.h': No such file or directory [C:\PathToMyProject\node_modules\node-librealsense\build\node_libr 
ealsense.vcxproj]```

Attempt 2

  * Added "node-librealsense" to package.json manually (using the version that the nodejs wrapper has in *its* package.json) then running `npm i --vs-configuration=Release` (this just produces the same error as I originally got when running `npm i node-librealsense`

I don't really understand how the comment you linked is meant to help (I mean that genuinely, not in a snarky way!) - how would running a install command in the wrappers/nodejs folder do anything to import the package into my project folder? Am I just misunderstanding something fundamental about how to install node-librealsense?

MartyG-RealSense, 2021-04-19:
The Nodejs installation instructions are admittedly complex.  It is probably best to follow the sequence of instructions in the **prerequisites** link below, and then enter the **npm install --save node-librealsense** command only once the list of prerequisite installation steps has been completed.

https://www.npmjs.com/package/node-librealsense#1-install-prerequisites

Comparing those instructions to your steps, it seems as though you may be missing This step:

**1.  Install Python 2.7.xx, make sure "Add python.exe to Path" is checked during the installation.**

MartyG-RealSense, 2021-04-26:
Hi @lachlansleight  Do you require further assistance with this case, please?  Thanks!

lachlansleight, 2021-04-28:
Hi! Yes I suspect I will,I just have an intermediate deadline coming up for this project that I don't need the real sense camera working for so this task is on hold until next week :)

MartyG-RealSense, 2021-04-28:
Thanks very much @lachlansleight for the update.  I will keep this case open for a further time period.  Good luck!

closed 2021-05-15 by MartyG-RealSense