realsenseai/librealsense #8848: reset D435 many times and python.exe stop

donaldlee2008, 2021-04-17:
reset D435 many times and python.exe stop 

             
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (7)|
                                     | 
| Platform                                 | PC
| SDK Version                            |  {2.43.0.3013 }                          | 
| Language                            |python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
 
Problem Event Name:	BEX64
  Application Name:	python.exe
  Application Version:	3.6.5150.1013
  Application Timestamp:	5abd3210
  Fault Module Name:	pyrealsense2.cp36-win_amd64.pyd
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	60570ac9
  Exception Offset:	00000000008ce27c
 Exception Code:	c0000409
  Exception Data:	0000000000000000
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	2052
 Additional Information  1:	c096
  Additional Information 2:	c096309941b54774d85a1abaa5ce91ee
  Additional Information3:	307c
  Additional Information 4:	307c6763608018aae047ae25ce88a9f7

here is the code
for i in range(2000) :
        ctx = rs.context()
        ctx.query_devices()[0].hardware_reset()
        time.slppe(10)
        pipeline = rs.pipeline()
        config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)#435
        config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30) 
        profile = pipeline.start(config)
        depth_sensor = profile.get_device().first_depth_sensor()
        depth_scale = depth_sensor.get_depth_scale()
        print("Depth Scale is: " , depth_scale)
        while True:
               # Get frameset of color and depth  
               for x in range(50):# Skip 5 first frames to give the Auto-Exposure time to adjust
                     frames =pipeline.wait_for_frames()
            time.sleep(2)
            break # break while to continue for  i

MartyG-RealSense, 2021-04-17:
Hi @donaldlee2008  I note that you are using Windows 7.  Could you use the information in the link below to check whether your Windows 7 is the 32-bit or 64-bit version please?

https://support.microsoft.com/en-us/office/determine-whether-your-computer-is-running-a-32-bit-version-or-64-bit-version-of-the-windows-operating-system-aac162a1-0cb3-46f2-888f-2f22897396ce#__toc289248465

donaldlee2008, 2021-04-18:
> Hi @donaldlee2008 I note that you are using Windows 7. Could you use the information in the link below to check whether your Windows 7 is the 32-bit or 64-bit version please? [...]

it is Windows 7   64-bit

MartyG-RealSense, 2021-04-18:
Okay thanks, I just wanted to make sure that you were using the correct pyrealsense2 pyd file (there is a 32-bit version of it too).

The title of your case is "reset D435 many times and python.exe stop".  Could you tell me more about what this means, please?  Does the program stop after you have started it and resetting the D435 does not fix it?   Or does the program work at first but stop working after you have reset the camera many times?

closed 2021-05-02 by MartyG-RealSense