realsenseai/librealsense #8849: fail to compile on Xubuntu with kernel 4.4.179-rk3399

w407022008, 2021-04-17:
Hi, @MartyG-RealSense 
With the instructions for [Linux installation](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md##prerequisites) I tried to compile on a Orange Pi 4B with XUbuntu 18.04 kernel version 4.4.179-rk3399 but failed in some sections

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400(D435i) } | 
| Operating System & Version |   {Linux (Ubuntu 18)}   | 
| Kernel Version (Linux Only)    |  4.4.179-rk3399                    | 
| Platform                                 | Orange Pi 4B  |
| SDK Version                            |  { (v2.43.0)}                          | 

### Issue Description

when `Using pre-build packages,` i could not add the repo because it dosen't support architecture 'armhf'.

> N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'armhf'

when `Configuring and building from the source code`,  I could not install it by using `./scripts/patch-realsense-ubuntu-lts.sh`, because it shows me that it dosen't support the kernel version 4.4.179-rk3399.

> Unsupported kernel version 4.4.179-rk3399 . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]

Is there any way?

closed 2021-04-17 by w407022008

vbgfbft, 2024-09-19:
How u finish it, I have same problem bro

vbgfbft, 2024-09-19:
my device is orangepi5plus, have same error with u

MartyG-RealSense, 2024-09-20:
Hi @vbgfbft  For Arm-based devices such as your Orange Pi, I recommend building the librealsense SDK from source code with the **libuvc backend** procedure at the link below, which builds using a pre-made build script called `libuvc_installation.sh`

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

w407022008, 2024-09-30:
Hi @vbgfbft . Let me know if you refer to [this](https://github.com/w407022008/DroneSys/blob/fd95dabcfcac86720f281403d31def82973169c4/README.md?plain=1#L65-L84).

MartyG-RealSense, 2024-09-30:
Hi @vbgfbft  Were you able to try my suggestion above about using the libuvc backend installation method, please?