realsenseai/librealsense #8850: Failed to compile on XUbuntu 18.04 kernel 4.4.179-rk3399 with Orange Pi 4B

w407022008, 2021-04-17:
Hi, @MartyG-RealSense 
With the instructions for [Linux installation](https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md##prerequisites) I tried to compile on a Orange Pi 4B with XUbuntu 18.04 kernel version 4.4.179-rk3399 but failed in some sections

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400(D435i) } | 
| Operating System & Version |   {Linux (Ubuntu 18)}   | 
| Kernel Version (Linux Only)    |  4.4.179-rk3399                    | 
| Platform                                 | Orange Pi 4B  |
| SDK Version                            |  { (v2.43.0)}                          | 

### Issue Description

when `Using pre-build packages,` i could not add the repo because it dosen't support architecture 'armhf'.

> N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://librealsense.intel.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'armhf'

when `Configuring and building from the source code`,  I could not install it by using `./scripts/patch-realsense-ubuntu-lts.sh`, because it shows me that it dosen't support the kernel version 4.4.179-rk3399.

> Unsupported kernel version 4.4.179-rk3399 . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]

Is there any way?

MartyG-RealSense, 2021-04-17:
Hi @w407022008   The librealsense Debian installation packages are designed for x86 / x64 processors and so will not work with devices that have an Arm processor such as Orange Pi (and also Raspberry Pi and Jetson).

You can instead build librealsense from source code using CMake with the **RSUSB backend** installation method.  This method, which uses the build configuration flag **-DFORCE_RSUSB_BACKEND=true**, installs the SDK via an internet connection and is not dependent on Linux versions or kernel versions and does not require patching,  Instructions can be found in the link below

 -DFORCE_RSUSB_BACKEND=true

w407022008, 2021-04-17:
Ok, Thanks,,,, I will have a try.. And any issues will be returned, if found.

w407022008, 2021-04-17:
Hi, @MartyG-RealSense . This way works on my platform, but when i `roslaunch realsense2_camera rs_camera.launch `, i got an error that realsense ros is v2.2.23 which is compile with [librealsense v2.43.0](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.23) but the librealsense built is v2.44.0. Do we have a ros wrapper for librealsense v2.44.0?

MartyG-RealSense, 2021-04-17:
When librealsense is updated, the ROS wrapper has to be rebuilt too otherwise the message that you received is generated.

There is not a ROS wrapper version to match to 2.44.0 at the time of writing this, though one is usually released for each SDK version.  Ideally the RealSense ROS wrapper should be matched as closely as possible to the librealsense version recommended in the wrapper release notes for that particular wrapper.  

You could build librealsense 2.43.0 with RSUSB using the source-code file for that version and then build ROS wrapper 2.2.23 again.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.23

MartyG-RealSense, 2021-04-24:
Do you require further assistance with this case, please?  Thanks!

w407022008, 2021-04-25:
Hi, yes, It works for me.

Steps:
- sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
- git clone https://github.com/IntelRealSense/librealsense.git
- sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at
- ./scripts/setup_udev_rules.sh
- mkdir build && cd build
- cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
- sudo make uninstall && make clean && make && sudo make install
- realsense_viwer

closed 2021-04-25 by w407022008

MartyG-RealSense, 2021-04-25:
Great to hear about your success - thanks very much for the update and sharing the method that worked for you!